nRF24/RF24Network #41: RF24Network_config.h owerwrites RF24_config.h

prophet-roshak, 2015-02-16:
Hello!

I'm currently my build environment with updated lib versions, and have encountered a problem compiling RF24Network ontop of new 'arch' model. 

RF24Network_config.h has lines 
```
#ifndef __RF24_CONFIG_H__
#define __RF24_CONFIG_H__
[skipped]
#endif
```
which effectively makes any arch redefinitions made in actual RF24_config.h unused while compiling RF24Network. Can macros defined in these lines be defined globally? or can actual RF24_config be included where? I'm willing to restructure config file, but I'll need some guides to save consistency across all platforms.

TMRh20, 2015-02-16:
Well, this is kind of just bad design. It was a quick fix to prevent issues when including *.h files in the 'wrong' order.

I think it could just be replaced with the following, but need to test:
https://gist.github.com/TMRh20/1e67b372aed37e19a4c8#file-rf24network_config-h

prophet-roshak, 2015-02-16:
Works perfectly, thanks :)

closed 2015-02-16 by prophet-roshak