nRF24/RF24Network #42: Adding the BBB to the RF24Network

desertgourd, 2015-03-04:
I have started to take a look at the RF24Network library and it looks like you need to make some of the type of changes you made to the RF24 library when you added the Arch directory.  Is this something you can do and what sort of time frame would it take.  I am interested in it but do not want to push you as I can get by for now with the RF24 library.

TMRh20, 2015-03-04:
Well, with RF24Network I've pretty much had an 'unofficial' change freeze
on the master branch, and have been working exclusively out of the [Dev](
https://github.com/TMRh20/RF24Network/tree/Development) branch, which I
thought should have the required includes etc. My plan is to 'archive' the
master branch and replace it with the resulting code from DEV, since it
should maintain compatibility, with all the extra features as optional (by
#define).

With all of the GPIO, SPI and timing functions handled in the core driver,
there should (theoretically) be very little that needs to be done in any of
the higher layers to support additional hardware. I've also updated the
Makefiles from RF24Network to RF24toTUN, so please post any issues you
encounter, and we can try to fix them.

TMRh20

desertgourd, 2015-03-04:
I am using the Development branch.
Specifically I tried to make the examples in the /examples_RPi directory and got the errors:
root@beaglebone:/home/rnh/rf24libs/RF24Network/examples_RPi# make
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -Wall -I../ -lrf24-bcm -lrf24network rx-test.cpp -o rx-test
In file included from /usr/include/stdio.h:927:0,
                 from /usr/local/include/RF24/arch/MRAA/mraa/pwm.h:43,
                 from /usr/local/include/RF24/arch/MRAA/mraa.h:30,
                 from /usr/local/include/RF24/arch/MRAA/RF24_arch_config.h:4,
                 from /usr/local/include/RF24/arch/includes.h:8,
                 from /usr/local/include/RF24/RF24_config.h:36,
                 from /usr/local/include/RF24/RF24.h:18,
                 from rx-test.cpp:4:
/usr/include/arm-linux-gnueabihf/bits/stdio.h: In function 'int vprintf(const char*, __gnuc_va_list)':
/usr/include/arm-linux-gnueabihf/bits/stdio.h:37:58: sorry, unimplemented: Thumb-1 hard-float VFP ABI
rx-test.cpp: At global scope:
rx-test.cpp:26:12: error: 'RPI_V2_GPIO_P1_15' was not declared in this scope
rx-test.cpp:26:31: error: 'BCM2835_SPI_CS1' was not declared in this scope
rx-test.cpp:26:48: error: 'BCM2835_SPI_SPEED_8MHZ' was not declared in this scope
make: *** [rx-test] Error 1

I modified the examples to use the 'RF24 radio(59,0);' command as in the RF24 examples.

TMRh20, 2015-03-04:
Are you sure you didn't forget the rx-test.cpp example? It's also trying to
compile using the RPi flags so you might need to update to get the latest
makefile.
If you look closely at the error output it will usually lead you to the
offending file and line number, in this case line 26 of rx-test.cpp. It
might look like a mess of info but it will usually give all the info needed
to find the problem.

desertgourd, 2015-03-04:
The first time through I did for get to update rx-test.cpp with  'RF24 radio(59,0);'  but the next time I added it and on make I got the same errors.  Also Made the RF24 library with 'make install -B' to use the SPIDEV libraries instead of MRAA and that got errors when making the examples:

root@beaglebone:/home/rnh/rf24libs/RF24Network/examples_RPi# make
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -Wall -I../ -lrf24-bcm -lrf24network rx-test.cpp -o rx-test
In file included from /usr/include/c++/4.6/bits/char_traits.h:41:0,
                 from /usr/include/c++/4.6/string:42,
                 from /usr/local/include/RF24/arch/BBB/spi.h:26,
                 from /usr/local/include/RF24/arch/BBB/RF24_arch_config.h:16,
                 from /usr/local/include/RF24/arch/includes.h:6,
                 from /usr/local/include/RF24/RF24_config.h:36,
                 from /usr/local/include/RF24/RF24.h:18,
                 from rx-test.cpp:4:
/usr/include/c++/4.6/bits/stl_algobase.h: In function 'int std::__lg(int)':
/usr/include/c++/4.6/bits/stl_algobase.h:981:15: sorry, unimplemented: Thumb-1 hard-float VFP ABI
make: *** [rx-test] Error 1

TMRh20, 2015-03-04:
Do you see a line in the makefile with ' cat /proc/cpuinfo ' ? If not it
needs to be updated, but if so then comment out the ccflags line. Also plz
post the results of running /cat/proc/cpuinfo  if so.

desertgourd, 2015-03-04:
I am not sure at this time if the rx-test.cpp file did get updated. so I updated again and ran the make on the examples_RPi directory.  I now get:

root@beaglebone:/home/rnh/rf24libs/RF24Network/examples_RPi# make
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -Wall -I../ -lrf24-bcm -lrf24network rx-test.cpp -o rx-test
In file included from /usr/include/stdio.h:927:0,
                 from /usr/local/include/RF24/arch/MRAA/mraa/pwm.h:43,
                 from /usr/local/include/RF24/arch/MRAA/mraa.h:30,
                 from /usr/local/include/RF24/arch/MRAA/RF24_arch_config.h:4,
                 from /usr/local/include/RF24/arch/includes.h:8,
                 from /usr/local/include/RF24/RF24_config.h:36,
                 from /usr/local/include/RF24/RF24.h:18,
                 from rx-test.cpp:4:
/usr/include/arm-linux-gnueabihf/bits/stdio.h: In function 'int vprintf(const char*, __gnuc_va_list)':
/usr/include/arm-linux-gnueabihf/bits/stdio.h:37:58: sorry, unimplemented: Thumb-1 hard-float VFP ABI
make: *** [rx-test] Error 1

This is different then the first errors on the make examples_RPi.  Line 4 of rx-test.cpp is:
```
#include <RF24/RF24.h>
```
It appears that It can't find the RF24.h file.

desertgourd, 2015-03-04:
RF24Network/examples_RPI/makefile did not have the  ' cat /proc/cpuinfo '  line.  I commented out the ccflags line and it ran make with on errors.  I will try running the helloworld_tx and rx.cpp examples.

desertgourd, 2015-03-04:
helloworld_tx and rx .cpp are both working.  It appears that I had not updated the rx-test.cpp and the RF24Network/examples_RPI/makefile needs the ' cat /proc/cpuinfo ' line update.  Thanks for the help.

TMRh20, 2015-03-04:
Thanks again for all your input/feedback/testing/troubleshooting, etc.,

closed 2015-08-29 by TMRh20