nRF24/RF24Network #43: [question] Any luck to print radio details?

chrisvel, 2015-03-24:
In nrf24 library there was a method radio.printDetails(); which used to print some useful information in the serial interface. What about RF24Network library? I couldn't find this anywhere in the source but it doesn't display an error when compiling.

spaniakos, 2015-03-24:
the method is still in the RF24 library.
the RF24 library is incuded via #include in the RF24Network_config.h, therefore there is no need for a new printDetails(); method. Moreover as arduino required the library in included in the main program as well, so feel free to use the radio.printDetails(); method.

chrisvel, 2015-03-24:
To be honest it doesn't print out anything and it's the only thing that doesn't work. Also I couldn't find it in any of the examples that's why I wondered If it exists.

spaniakos, 2015-03-24:
what board are you using ? 
since the printDetails(); method uses printf_P or printf calls for all boards (RPI and Arduino), you need to include printf.h and call printf_begin(); method in order to redirect printf to Serial output.

chrisvel, 2015-03-24:
A chinese arduino nano (!). Actually there was printf_begin(); missing and I added it. Also #include "printf.h" was missing. So everything finally worked ok. Thanks !

closed 2015-05-07 by TMRh20