nRF24/RF24Network #44: Python wrapper: "UnicodeDecodeError" when reading binary data

joernesdohr, 2015-03-31:
I am using the RF24 Python wrapper with Python 3.2, sending binary data between my RPi and Arduino. Data transmission from RPi to Arduino works flawlessly, but transmitting numbers, e.g. signed shorts, as binary data result in following error when they include a hexadecimal pair greater than  0x7f in value:

    File "/usr/local/lib/python3.2/dist-packages/aurora/controller/hardware/rf24controller.py", line 49, in loop
         header, payload = self.network.read(30)
    UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte

The above error is given when trying to read a short of 127 in value.

Correct me if I'm wrong, but after glancing over it, my guess would be that the issue is caused by using a char array as a buffer in the Python wrapper for the network.read() method.

mz-fuzzy, 2015-03-31:
I cannot see such behaviour with Python2. On RPi I have troubles to install libs to Python3 because python-setuptools on raspbian are older and not supporting python3 yet. So it would be nice if you can reproduce this with python2 and check whether the issue is Python3 specific. Also publishing code on both transmitting and receiving side would help.

joernesdohr, 2015-04-01:
Ok, I just installed libboost-python1.49-dev, the wrappers for python 2.7 and run the helloworld_rx.py example, which works just fine with any value.

The transmission was as simple as:

	short num = 128;
	RF24NetworkHeader response_header(target_node, 6);
	network.write(response_header, &num, sizeof(num));

Unfortunately, I'm now receiving another cryptic library error after having rebuilt the RF24Network wrapper with libboost-python3, which now prohibits me from importing the RF24Network modules and running my application:

	Traceback (most recent call last):
	  File "/usr/local/lib/python3.2/dist-packages/aurora/controller/aurora_controller.py", line 34, in __init__
		from aurora.controller.hardware.rf24controller import RF24Controller
	  File "/usr/local/lib/python3.2/dist-packages/aurora/controller/hardware/rf24controller.py", line 8, in <module>
		from RF24Network import *
	ImportError: /usr/local/lib/python3.2/dist-packages/RF24Network.cpython-32mu.so: undefined symbol: _ZN11RF24Network9availableEv

Do you have any idea what the undefined symbol is about? I used the following instruction to build the wrapper, which should also be the same as the one I used successfully a month ago (probably not the most elegant of solutions...):

	sudo g++ pyRF24Network.cpp -I/usr/include/python3.2 -I/usr/local/include -I/home/pi/workspace/auroraproject/RF24Network/RPi -lboost_python3 -o /usr/local/lib/python3.2/dist-packages/RF24Network.cpython-32mu.so -shared

Either way, this is the code I was running in my Python 3 application. The network.update() and network.read() were called in the loop() method, running in a separate thread:

	class RF24Controller(HardwareController):
		def __init__(self):
			self.device_id = 0o1

			# radio setup for RPi B Rev2: CS0=Pin 24
			self.radio = RF24(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ)
			self.network = RF24Network(self.radio)
			
		def begin(self):
			logging.info("Starting RF24 network...")
			self.radio.begin()
			time.sleep(0.1);

			self.network.begin(90, self.device_id)
			self.radio.printDetails()

			self.io_thread = threading.Thread(target=self.loop)
			self.io_thread.start()
			
		def loop(self):
			"""This loop is run in a separate thread and handles any incoming messages from network devices.
			"""
			while 1:
				self.network.update()
				while self.network.available():
					header, payload = self.network.read(30)
					logging.info("Received device message of type '%s'" %header.type)

					if header.type == DeviceMessageType.I_AM:
						logging.info('Received I_AM message')
						self.__process_i_am(header, payload)

Meanwhile, I'm trying to repair my RF24Network python 3 library to continue troubleshooting.

joernesdohr, 2015-04-01:
Dangit, the gcc instructions were simply missing `-lrf24network`...

I just ran the helloworld_rx.py example in Python 3.2 with only fixed print method and new octet notation and too am receiving the UnicodeDecodeError. As mentioned previously, the example runs fine in Python 2.7. So it really seems like a Python 3 specific error.

mz-fuzzy, 2015-04-01:
Thanks for analysis @joernesdohr.

To make the issue clear for everyone: there is no support of python3 in RF24 and RF24Network libs as of now. Python2 and Python3 are different and incompatible languages. Beside the string issue, there might be additional ones, for example with installation scripts on RPi. So Python3 support needs to be handled as a new feature there.

@joernesdohr, I'd be glad if you explore aspects of Python3 support in RF libs and share findings. For me it will take a while to find a time for that.

joernesdohr, 2015-04-02:
Python 3 support would be damn sweet, but one can't have everything right away! I'll be glad to contribute as much as I can in that regard.

And sweet baby jesus! In a last-ditch effort I actually managed to forward the C++ byte buffer as a MemoryView object to Python. My read_wrap method looks like this now:

	bp::tuple read_wrap(RF24Network& ref, size_t maxlen)
	{
		size_t len;
		char *buf = new char[maxlen+1];
		RF24NetworkHeader header;

		len=ref.read(header, buf, maxlen);
		// std::string str(buf, len);

		Py_buffer buffer;
		int result = PyBuffer_FillInfo(&buffer, 0, buf, len, true, PyBUF_CONTIG_RO);
		if (result == -1)
		{
				PyErr_Print();
				exit(EXIT_FAILURE);
		}
		bp::object memoryView(bp::handle<>(PyMemoryView_FromBuffer(&buffer)));
		// delete[] buf;
		return bp::make_tuple(header, memoryView);
	}

I'm not sure if that is causing memory leaks though, as I had to comment out the deletion of the `buf` array. Anyway, at least I'm no longer getting garbage data and can successfully unpack the byte data as a struct, which looks like this for a struct with 5 short variables:

    header, payload = self.network.read(10)
    unpack('<hhhhh', bytes(payload)[0:calcsize('<hhhhh')])

joernesdohr, 2015-11-14:
I took your changes from https://github.com/TMRh20/RF24/issues/164 and updated my pyRF24Network Python 3.2 version file and it seems to work fine: https://github.com/joernesdohr/RF24Network/commits/master/RPi/pyRF24Network/pyRF24Network.python3.2.cpp

I tried to run the helloworld_rx.py / _tx.py examples as well, between an Arduino and RPi, but noticed that the struct information is incorrectly described as "<qi" instead of "<LL" for two unsigned Long variables. The example works with the RPi as receiver, but for some reason I get a

    struct.error: argument out of range

when using "<LL" format for helloworld_tx. Going with the original "<qi" results in a 

   RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.

Even though I can send byte data just fine with my remote light control application. The RF24 radio is configured the same, so that's not the issue...

mz-fuzzy, 2015-11-14:
> It seems to work fine with my own light application, but I have trouble running the helloworld_rx/tx example, between an Arduino and RPi. I noticed that the struct information is incorrectly described as "<qi" instead of "<LL" for two unsigned Long variables. The example works with the RPi as receiver, but for some reason I get a

> struct.error: argument out of range
> when using "<LL" format for helloworld_tx.

Now I recall why I put <qi into the examples; the millis function as it's written exceeds range of long. So the examples need to adapt in order to be compatible with C-RPi and Arduino (for 2 longs) and also the millis function must produce value in 'long' range.

For the wrapper adaptation - I was going to start that as well, but if you are close already - go forward with a pull request once finished, I can review that then.

joernesdohr, 2015-11-23:
I think I got it working, took me a while to test because only later I realized that my libboost-python for 2.7 was faulty. You can check my changes here: https://github.com/joernesdohr/RF24Network

The helloworld examples work fine for Python 2.7 (for _tx I just sent packets_sent integer twice since pack() is really finicky about millis() size...). The thing to note though is that now network.read() returns a bytearray object, so for Python 2.7 you need to turn it into a String for unpack -- or use unpack_from(format, buffer(payload)). Not sure what the wanted behaviour is here.

The example helloworld_rx.python3.py for Python 3 works as well, just helloworld_tx.python3.py continues to return the "RF24 HARDWARE FAIL" message while eternally trying to resend the first message. Other than the payload, it doesn't differ from my working application. Maybe it works on your setup?

joernesdohr, 2015-11-23:
Created a pull request: https://github.com/TMRh20/RF24Network/pull/68

mz-fuzzy, 2015-11-23:
unfortunately I'm not able to test on RPi for now. @TMRh20 or someone else, can you check/test https://github.com/TMRh20/RF24Network/pull/68 especially for the error @joernesdohr is getting?

joernesdohr, 2015-11-23:
My RPi can send messages fine with the latest changes. But if I remove the newly added line

    from __future__ import print_function

I receive the aforementioned Hardware Failure error. Any idea why that is?

TMRh20, 2015-11-26:
Testing using @joernesdohr fork seems to work fine using ``` sudo ./helloworld_tx ``` or rx but running via ```sudo python3 ./helloworld_tx``` or rx causes an immediate exit from the program. 

The tx sketch exits immediately after sending, although the transmission seems ok.
The rx sketch exits immediately after receiving the second payload.

Removing the network.reads/writes results in no exit from the program.

I'm finishing up a clean install on another RPi, so will test on that soon also.

TMRh20, 2015-11-26:
Update: On the fresh install of Raspbian Jessie I did the following:

Building RF24:
1. Got error Virtual memory exhausted: Cannot allocate memory
   Increased [swap file size](https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/)
2. sudo ln -s /usr/lib/arm-linux-gnueabihf/libboost_python-py34.so.1.55.0 /usr/lib/arm-linux-gnueabihf/libboost_python3.so

python3 ./setup.py build
sudo python3 ./setup.py install

python ./setup.py build
sudo python ./setup.py install

All seems to be working fine, again regardless of the __future__ import

Not sure what is wrong with the first RPi, but I'm assuming its something I did...

joernesdohr, 2015-12-08:
Thanks for the merge @TMRh20! Looks like this can finally be closed.

closed 2015-12-08 by joernesdohr