nRF24/RF24Network #45: Trouble with sleep mode (master branch)

myersjj, 2015-04-26:
On an Arduino Pro Mini 3.3v, sleep mode works fine, but on a bare Atmega328 on 3.3v power on breadboard it will never go to sleep.  I don't see any substantive difference in the sleep mode code in the dev branch.  

Any ideas on what could be wrong or how to debug this?  Thanks!  Great libraries!

myersjj, 2015-04-26:
I merged in the sleep mode changes from the dev branch, but no change - it still never sleeps :(

myersjj, 2015-04-26:
2 other hardware differences, the pro mini is running on 3.3v USB power. The breadboard atmega328 is using batteries and the internal 8Mhz oscillator.

myersjj, 2015-04-27:
After adding your code to return !wasInterrupted from network.sleepNode, it returns 0 (false) which means it was interrupted. So what else could be coming in as interrupt 0 (pin 2)?

TMRh20, 2015-04-27:
If pin 2 is floating, you could try tying it high via a resistor, or setting pinmode to INPUT_PULLUP.

myersjj, 2015-04-27:
I have pin 2 connected to the nrf24L01 IRQ pin (on both devices). I really need the sleep mode to handle  interrupts. I will try it without the interrupts though just for grins.

myersjj, 2015-04-27:
OK, without an interrupt pin specified like: network.sleepNode(4, 255) it sleeps ok, So it seems interrupts are coming in from somewhere that are NOT occurring on the Pro Mini.

TMRh20, 2015-04-27:
Hmm it would seem that the nrf pin is being detected as low. It might be
related to how it is being powered.

myersjj, 2015-04-27:
ok, i just checked the nrf irq pin (pin 2 on 328) and it is indeed low. So I guess I need to try the INPUT_PULLUP...  Well I tried that to no avail. Pin 2 now has 1.2v but still interrupting...  Any suggestions??

TMRh20, 2015-04-27:
That is the problem. Typically need 0.6-0.7vcc for a pin to read high.

myersjj, 2015-04-27:
So why is this different on a bareboard atmega328 and an Arduino Mini Pro I wonder?  Also, when pin 2 read 0v it didn't work and when it reads 1.2v it still doesn't work. So I don't understand the .6-.7 statement :(

TMRh20, 2015-04-27:
http://www.atmel.com/dyn/resources/prod_documents/doc8161.pdf

http://www.omega.com/literature/transactions/volume2/digitalio.html

http://en.wikipedia.org/wiki/Digital_signal

myersjj, 2015-04-28:
I changed this line in RF24Network.cpp and it works now:

attachInterrupt(interruptPin,wakeUp, FALLING);  // does not work if LOW used here

TMRh20, 2015-05-07:
I assume this has to do with the power supply but am not sure. With my previous comment, I meant that the pin voltage generally needs to be greater than about 1/2 or 0.5 VCC to be recognized as HIGH. If you are reading 1.2V and supplying 3.3V, the pin is LOW.

I've noticed similar problems between using LOW and FALLING interrupts with nrf modules in the past, but haven't really been able to look into it much.

emax9, 2015-08-03:
For the example of Network_Ping_Sleep of Newly Optimized RF24Network Layer, the nodes other than base node never comes out of the sleep with NRF24L01 IRQ because the radio pin is never triggered even when base node sends message and this happens because of radio.stopListening(); before going to sleep in the following function. However, it does awake once the 8 cycles of 1 second interval set by watchdog timer experies.

/*** CALLING THE NEW SLEEP FUNCTION ***/    
  if(millis() - sleepTimer > awakeTime&& NODE_ADDRESS){  
     Serial.println("Sleep");
     sleepTimer = millis();
     delay(100);
     radio.stopListening();
     network.sleepNode(8,0);
     Serial.println("Awake"); 
  }

If I comment out radio.stopListening(); then it works fine. 
If the radio stops listening and goes to sleep, how it can generate IRQ for not receiving a message? Please correct if I am wrong. BTW I have tested this code on Arduino Uno. 

Thanks.

TMRh20, 2015-08-29:
@emax9 Sorry for the delayed reply, but yes, this is an old mistake in the example that should have been corrected. (I was experimenting with ack-payloads waking a device) The radio should not stop listening as you have identified.

emax9, 2015-09-12:
@TMRh20 thanks for the reply and clarification...
have you managed to get the LOW POWER feature using sleep mode working?
I am also banging my head to get it work... my problem is that ATmeag328 never goes to power-down sleep-mode as long as radio is powered-up and remains connected to SPI. If I somehow remove radio pins from MCU it does sleep with power-down. 
Maybe it is the SPI clock or MOSI/MISO data which don't let the MCU power-down enable... but not sure. As for CS and CSN pins they should remain active for the MCU to wake from from receiver IRQ.

TMRh20, 2015-09-13:
I've never noticed a problem with the MCU not powering down. How do you draw this conclusion?

>As for CE and CSN pins of radio, they should remain active for the MCU to wake from receiver IRQ.
Only the interrupt pin is need to wake the MCU.

emax9, 2015-09-17:
Sorry, the MCU powers down successfully... I was mistaken by the total power consumption with the following break-down:

MCU Sleep
---------------
MCU current: 0.06 mA
Radio current (in receive mode): 15 mA
3.6V to 5V step-up converter current: 15 mA
Total current: 30 mA

MCU Awake
---------------
MCU current: 14 mA
Radio current (in receive or transmit mode): 15 mA
3.6V to 5V step-up converter current: 15 mA
Total current: 44 mA

The step-up converter is taking too much current which I am looking to reduce somehow.

closed 2015-12-26 by Avamander