nRF24/RF24Network #47: RF24 between RPi and Arduino

Aterocana, 2015-05-13:
I'm running this code in the RPi:
```
#include <cstdlib>
#include <iostream>
#include <RF24/RF24.h>
#include <RF24Network/RF24Network.h>
#include <ctime>
#include <stdio.h>
#include <time.h>

using namespace std;

/*
	IMPORTANTE: IL PRIMO PARAMETRO CAMBIA A SECONDA DELL'ARCHITETTURA:
	Raspberry Pi B V2 	- CE sul pin 15 (GPIO22): RPI_V2_GPIO_P1_15
	Rasbperry Pi B+ 	- CE sul pin 22 (GPIO25): RPI_BPLUS_GPIO_J8_22
*/	
RF24 radio(RPI_BPLUS_GPIO_J8_22, BCM2835_SPI_CS0, BCM2835_SPI_SPEED_8MHZ);  

RF24Network network(radio);

// Address of our node in Octal format (01,021, etc)
const uint16_t this_node = 00;

// Address of the other node
const uint16_t other_node = 01;

struct payload_t {
    unsigned long request;            //tipo di richiesta 0= ASK, 1=DO
    unsigned long op;                        //ON=1, OFF=0, STATUS=2
    unsigned int pin;    
};

int main(int argc, char** argv) 
{
	// Refer to RF24.h or nRF24L01 DS for settings

	radio.begin();
	
	delay(5);
	network.begin(/*channel*/ 90, /*node address*/ this_node);
	//radio.printDetails();	

	network.update();	
	printf("Sending ..\n");
	payload_t payload = { 1, 54, 2};
	RF24NetworkHeader header(/*to node*/ other_node);

	bool ok = network.write(header,&payload,sizeof(payload));

	if (ok){
		printf("ok.\n");
		return 0;
	}
	else{ 
		printf("failed.\n");
		return 1;
	}
}
```

and this one on Arduino:
```
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(9,10);                // nRF24L01(+) radio attached using Getting Started board 

RF24Network network(radio);      // Network uses that radio
const uint16_t this_node = 01;    // Address of our node in Octal format ( 04,031, etc)
const uint16_t other_node = 00;   // Address of the other node in Octal format

struct payload_t {                 // Structure of our payload
  unsigned long request;
  unsigned long op;
  unsigned int pin;
};

void setup(void)
{
  Serial.begin(57600);
  Serial.print("receiving at node ");
  Serial.println(this_node); 
  SPI.begin();
  radio.begin();
  network.begin(/*channel*/ 90, /*node address*/ this_node);
}

void loop(void){
  
  network.update();                  // Check the network regularly

  
  while ( network.available() ) {     // Is there anything ready for us?
    
    RF24NetworkHeader header;        // If so, grab it and print it out
    payload_t payload;
    network.read(header,&payload,sizeof(payload));
    Serial.print("Received packet [req: ");
    Serial.print(payload.request);
    Serial.print(",op: ");
    Serial.print(payload.op);
    Serial.print(",pin: ");
    Serial.print(payload.pin);
    Serial.println("]");
  }
}
```

Everything is working since I get, on the Arduino receiver: 
```
Received packet [req: 1,op: 54,pin: 2]) 
```

If I change, in both the transmitter and the receiver, the payload structure according to this:
```
struct payload_t {
    unsigned int request;            //tipo di richiesta 0= ASK, 1=DO
    unsigned int op;                        //ON=1, OFF=0, STATUS=2
    unsigned int pin;    
};
```

what I get from the Arduino receiver is:
```
Received packet [req: 1,op: 0,pin: 54]
```

which is wrong since I'm sending {1, 54, 2}. Basically changing fields from long to int seems to invalidate the transmission soundness.
Any clues?

Aterocana, 2015-05-13:
By the way, between 2 arduinos everything works smoothly (off course, writing an ad hoc code for the sender Arduino).

TMRh20, 2015-05-13:
http://electronics.stackexchange.com/questions/106933/how-different-are-8-bit-microcontrollers-from-32-bit-microcontrollers-when-it-co

http://en.wikipedia.org/wiki/Data_structure_alignment

http://www.youtube.com/watch?v=FClGhto1vIg

Aterocana, 2015-05-14:
I'm far from being an expert so thanks for teaching me something new. I'm reading your links.

Aterocana, 2015-05-14:
I found that sizes in RPi (32 bit register, so 4 bytes) are:
char -> 1 byte,
int -> 4 bytes,
long -> 4 bytes

while in Arduino (8 bit register, so 1 bytes)
char -> 1 byte,
int -> 2 bytes,
long -> 4 bytes

According to this I can use long and char with no problems. If I use an int I should provide the proper padding. Example:
I define this structure in 32 bit architecture:
```
struct payload_t {
	char request; 	//tipo di richiesta: 0=ask, 1=do

	//campi do
  	char op;			//0=OFF, 1=ON, 2=STATUS
  	int pin;  	

  	//campi ask
  	char type; 		//tipo di dispositivo
}
```

Which means that I have, in my 4 bytes:
request (1 byte) - op (1 byte) - padding (2 bytes)
pin (2 bytes) - padding (2 bytes)
type (1 byte)

If I use the same structure in arduino, since its register is 1 byte, it would read the padding bytes as variables (pin would take the 2 padding bytes). According to this I should define a structure like this one:
```
struct payload_t {                 // Structure of our payload
    char request;            //tipo di richiesta 0= ASK, 1=DO

  // campi do
    char op;                        //ON=1, OFF=0, STATUS=2
    char padding[2];
    int pin;                    //PIN di destinazione    
    char second_padding[2];

 // campi ask
    char type;              //tipo di dispositivo
}
```
It seems it's working. Am I right or am I missing something else? Thanks.

closed 2015-08-29 by TMRh20