nRF24/RF24Network #48: Edit New issue Development - Corrupted message_buffer issue during fragmented packet duplicate disposal

brunnels, 2015-05-17:
I discoverd an issue on Arduino where a duplicate fragmented packet will corrupt the message_buffer.  I have an RF24 node that sends out a message to the base when it comes online.  Because of this the RF24NetworkHeader::id will always be the same: 2.  The message is always the same too.  If I reset the node a few times without sending out any additional fragmented messages the BASE will keep dropping the message.  The problem is that something gets messed up with the message_buffer and a portion of each message gets appended to the original each time until the BASE is restarted.  Here's an example:
```
1st:  Payload: rf24/5/online^node 5 online
2nd:  Payload: rf24/5/online^node 5 onlineine
3rd:  Payload: rf24/5/online^node 5 onlineineine
4th:  Payload: rf24/5/online^node 5 onlineineineine
```

If I split up the if statement and comment out this part at line 422 in RF24Network.cpp the issue goes away but I know the real bug is elsewhere.  I just haven't tracked it down yet.

```
(frag_queue.header.id == header->id && frag_queue.header.from_node == header->from_node)
```

TMRh20, 2015-05-18:
I haven't gotten a chance to look at this in detail, but I'm sure it has to do with the reassembly code. The line you point out also seems to confirm that too, so I think you are looking in the right place. I'll see about taking a closer look myself pretty soon.

TMRh20, 2015-06-17:
This should be fixed now, with sequentially received, duplicate fragmented payloads being dropped.

closed 2015-08-29 by TMRh20