nRF24/RF24Network #54: network.available() in rPi always false

chrisvel, 2015-08-18:
I can make two arduinos work together but not a rPi and an Arduino:

base.cpp
```
#include <RF24/RF24.h>
#include <RF24Network/RF24Network.h>
#include <iostream>
#include <ctime>
#include <stdio.h>
#include <time.h>

using namespace std;

//////////////////////////////////////
// Create the structures for data
//////////////////////////////////////
struct data_received {
  uint16_t node;
  uint16_t temp;
  uint16_t humi;
  uint16_t light;
  uint16_t door;
  uint16_t pir;
  uint32_t hkey;
};

struct data_to_send {
  uint32_t hkey;
}confirm;

unsigned long finish = 0;
unsigned long elapsed = 0;

//////////////////////////////////////
// Setup nRF24L01+ module & Oled
//////////////////////////////////////
RF24 radio(RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

RF24Network network(radio);
const uint64_t this_node  = 00; // base

//////////////////////////////////////
// loop
//////////////////////////////////////

int main(int argc, char** argv)
{

  //////////////////////////////////////
  // setup
  //////////////////////////////////////

  // Initialize interfaces
  cout<<"Starting interfaces...";
  radio.begin();
  radio.setRetries(7,7);
  delay(5);
  cout<<"complete]\n";
  network.begin(90, this_node);
  radio.setDataRate(RF24_250KBPS);
  radio.printDetails();

  while(1) {

    network.update();
    data_received payload;

    cout<<"**"<<network.available()<<'\n';

    // listen for incoming clients
    while ( network.available() )
      {
        RF24NetworkHeader header;
        confirm.hkey = 0;

        // read the next available header
        network.peek(header);

        // print NRF24 header data
        cout<<"\n[header info => id: ";
        cout<<header.id<<" type: ";
        cout<<header.type<<" from_node: ";
        cout<<header.from_node<<"]";

        // read data
        network.read( header, &payload, sizeof(payload) );

        // Print data received
        cout<<"[NODE: "<<payload.node<<" ]=>  ";
        cout<<"[temp: "<<payload.temp<<"*C] ";
        cout<<"[humi: "<<((float)payload.humi)<<"%] ";
        cout<<"[light: "<<((float)payload.light)<<"%] ";
        cout<<"[door: "<<payload.door<<"] ";
        cout<<"[pir: "<<payload.pir<<"] ";
        cout<<"[hkey: "<<payload.hkey<<"] ";

        // send back data to confirm integrity
        confirm.hkey = payload.hkey;
        cout<<"...";
        RF24NetworkHeader header2(header.from_node);

        // starts counting just before sending
        elapsed = 0;
        unsigned long start = millis();

        bool sent_back = network.write( header2, &confirm, sizeof(confirm));

        if (sent_back == true){
          finish = millis();
          elapsed = finish - start;
          cout<<elapsed;
          cout<<"ms";
          cout<<" - Confirmed back \n";
        }
        else{
          cout<<"Not confirmed";
        }

      }

      int value = 0;
      float vin = 0.0;
      value = payload.light;
      vin = 3.472 * value / 1024;
      cout<<"VIn: "<<vin<<'\n';

      cout<<"Listening for nodes.."<<'\n';
      delay(1000);
  }
  return 0;
}
```

a node (arduino)
```

#include <dht11.h>
#include "RF24Network.h"
#include "RF24.h"
#include <SPI.h>
#include "printf.h"
#include <JeeLib.h>

//////////////////////////////////////
// Create the structures for data
//////////////////////////////////////
struct data_to_send {
  uint16_t node;
  uint16_t temp;
  uint16_t humi;
  uint16_t light;
  uint16_t door;
  uint16_t pir;
  uint32_t hkey;
};

struct data_received {
  uint32_t hkey = 0;
};

//////////////////////////////////////
// Setup nRF24L01+ module
//////////////////////////////////////
RF24 radio(8,7);
RF24Network rf24Net(radio);
const uint64_t this_node =  02; // node A
const uint64_t other_node = 00; // base

//////////////////////////////////////
// Setup the rest of it
//////////////////////////////////////

const int led_A_Pin = 6;
const int optoPin   = A2;

unsigned long currentTime = 0;
unsigned long startTime   = 0;
bool runOnce = true;

ISR(WDT_vect) { Sleepy::watchdogEvent(); } // Setup the watchdog

//////////////////////////////////////
// setup
//////////////////////////////////////
void setup(void) {

  // Initialize interfaces
  printf_begin();
  Serial.begin(9600);
  Serial.print(F("Starting interfaces..."));
  SPI.begin();
  radio.begin();
  Serial.print(F("complete]\n"));
  rf24Net.begin(90, this_node);
  radio.setDataRate(RF24_250KBPS);
  radio.printDetails();
  analogReference(INTERNAL);

  // initialize pins
  pinMode(led_A_Pin, OUTPUT);

}

//////////////////////////////////////
// loop
//////////////////////////////////////
uint32_t txTimer = 0;

void loop(void)
{
    digitalWrite(led_A_Pin, HIGH);
    delay(200);
    digitalWrite(led_A_Pin, LOW);
    delay(200);
    digitalWrite(led_A_Pin, HIGH);
    
    currentTime = millis();

    if (currentTime - startTime <= 1000) {
      Serial.println(currentTime - startTime);
    }
    else {
       digitalWrite(led_A_Pin, LOW);
    }
  
  rf24Net.update();

  if (millis() - txTimer > 1000) {
    data_to_send payload;
    payload.node  = 02;
    payload.temp  = 11;
    payload.humi  = 11;
    
    payload.light = analogRead(A2);
    payload.door  = 0;
    payload.pir   = 0;
    payload.hkey  = random(10000000, 99999999);

    // starts counting
    unsigned long elapsed = 0;
    unsigned long start = millis();
    Serial.println(payload.hkey);

    digitalWrite(led_A_Pin, HIGH);
    delay(100);
    digitalWrite(led_A_Pin, LOW);
    delay(100);
    digitalWrite(led_A_Pin, HIGH);
    delay(100);
    digitalWrite(led_A_Pin, LOW);
    delay(100);

    // sends packet
    RF24NetworkHeader header(other_node);

    Serial.println("Sending data to base every 30 sec.");
    txTimer = millis();
    bool ok = rf24Net.write( header, &payload, sizeof(payload) );
    if (ok) {
      unsigned long finish = millis();
      elapsed = finish - start;
      Serial.println("ok.");
      Serial.print(elapsed);
      Serial.print("ms\n");
      rf24Net.update();
    }
    else {
      Serial.println("failed.");
    }
  }

  while ( rf24Net.available() )
  {
    Serial.println("Received packet");
    RF24NetworkHeader header2;
    data_received confirm;
    rf24Net.read(header2, &confirm, sizeof(confirm));
    Serial.println(confirm.hkey);
    Serial.println("------------------------------");
    
    Serial.println(F("Going to sleep..."));
    delay(100);
    radio.powerDown();
    Sleepy::loseSomeTime(2000);
    radio.powerUp();
    Serial.println(F("Woke up!"));
  }
}
```

Can you please help? Is it a bug or am I doing something wrong ?

chrisvel, 2015-08-19:
Also from cat /var/sys/log :

```
Aug 19 06:10:09 raspberrypi kernel: [  638.533222] spi_master spi0: DMA transfer timed out
Aug 19 06:10:09 raspberrypi kernel: [  638.533284] spi_master spi0: failed to transfer one message from queue
Aug 19 06:10:09 raspberrypi kernel: [  638.533364] fb_ili9340 spi0.0: fbtft_update_display: write_vmem failed to update display buffer
```

TMRh20, 2015-08-20:
Looks good to me. The RPi pinout is about as confusing as possible, so I would look [there first](http://pi.gadgetoid.com/pinout) . I noticed your Arduino has the default pins (7,8) reversed, but if it works Arduino-to-Arduino, probably not the issue. Make sure you have [spi enabled](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md) also on the RPi. I use RPi regularly with RF24, so should not be any obvious bugs.

chrisvel, 2015-08-20:
Thanks, this is the pinout I am using http://tmrh20.github.io/RF24/index.html. I just checked it again. This is also the line that connects the CE/CSN pins `RF24 radio(RPI_BPLUS_GPIO_J8_15,RPI_BPLUS_GPIO_J8_24, BCM2835_SPI_SPEED_8MHZ);`. The configurations: http://i.stack.imgur.com/itroc.png.

TMRh20, 2015-08-20:
Hmm, judging by the F0F0... addresses, looks like you have an old version of RF24Network lib in your Arduino library folder?  On windows, its ``` my documents/Arduino/Libraries ``` and you prob want to delete any RF24Network folders, then re-install for Arduino, from the same repo you installed on RPi.

chrisvel, 2015-08-20:
It's very strange. The pingpair_dyn works, the rPi receives the payload and displays it but the arduino transmits and returns "Failed". Are there any special settings I should look for ? (except the obvious channel etc. ones)

TMRh20, 2015-08-20:
No special settings, but some 'clone' nrf24l01 modules don't support dynamic payloads. 

See [here](http://hackaday.com/2015/02/23/nordic-nrf24l01-real-vs-fake/), [here](http://forum.mysensors.org/topic/1153/we-are-mostly-using-fake-nrf24l01-s-but-worse-fakes-are-emerging), and [here](http://sigrok.org/wiki/Protocol_decoder:Nrf24l01)

chrisvel, 2015-08-20:
This could make sense. Yes, I might have an old version for the arduino, although I believe I updated it recently. I 'll check it out again. What made you think that ? What about the F0F0 addresses ? The version on the rPi is the latest one. I am using a rPi B+.

chrisvel, 2015-08-20:
Could it be an issue with make ?

TMRh20, 2015-08-20:
> What made you think that ? What about the F0F0 addresses ?

The fact that my version of RF24Network uses different addressing. (using hex values 0xcc,0xc3,0x3c,0x33,0xce,0x3e,0xe3,0xec ) [specifically chosen for nrf24 radio transmission](http://maniacalbits.blogspot.ca/2013/04/rf24-addressing-nrf24l01-radios-require.html) .  Look at them in binary to understand why...

Most likely not a make issue, or it wouldn't compile probably...

chrisvel, 2015-08-20:
What would you suggest?

TMRh20, 2015-08-20:
Uhh.. the above :p  Basically check all your basics and proper library installation. This type of thing seems to usually be either something odd or something overlooked.

chrisvel, 2015-08-20:
I won't disagree with you.. I'll check it out and come back ! thanks

chrisvel, 2015-08-20:
Where should I see the difference of a new library in arduino ? in  TX_ADDR = 0xf0f0f0f069 ?

chrisvel, 2015-08-20:
Check this out also: output of pingpair_dyn http://i.stack.imgur.com/mhcmc.png

carrer, 2015-08-20:
Hmm... the [documentation](http://maniacbug.github.io/RF24/classRF24.html#a30a2733a3889bdc331fe2d2f4f0f7b39) says you cannot open two pipes at once (or you listen, or you write). I'm not sure how you're handling this, but maybe you're not.
Also, your modules can be "out-of-sync"... I mean, one starts listening when the other has already sent the data. Can't that be also the case?

chrisvel, 2015-08-20:
The pingpair_dyn and gettingstarted examples do not work which means that the communication fails. The good thing is that I saw 2 packets received oute of one thousand, so at least the connections are right. The examples should work out of the box with the minimal setup of the CE/CSN pins. 

Was it easy for you to connect them ?

carrer, 2015-08-20:
Actually I did it long time ago and I did not use RF24Network. I remember having this sort of problems as well, gotta pay attention on the time it takes for switching pipes/mode and not trying to hear something that has already been said. Have you tried to wait few milliseconds before answering/acknowledging the package straight away?

chrisvel, 2015-08-20:
This is what I also thought, but shouldn't the original examples work ok after you change the CE/CSN pins? Actually I am thinking to try your example from your blog also because it was working, right ? What concerns me most is the libraries incompatibilities that TMRh20 said, but I've already created new arduino and rPi installations from scratch. I also re-imaged my raspbian SD card. 

The last thing I was thinking is that I have the RPi connected to USB power. They say it needs 2A to work OK and I suppose that the USB offers 500mA, but why should it be If the nRF needs only some XXmA in order to send and receive data? 

Next thing will be to say a prayer.

TMRh20, 2015-08-21:
@chrisvel The examples are all thoroughly tested and working out of the box with nrf24l01 *compatible* modules. Judging by your pingpair_dyn example output, I'm assuming you have 'clone' modules that don't support dynamic payloads. If things test ok with the other RF24 core lib examples, than that probably is the case. 

With your initial example code, the printDetails() output on Arduino should look like this (note: the code works fine for me): 

```
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xcccccc3ccc 0xcccccc333c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccccc33
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x27
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x05
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

```

If it looks different, you need to make sure to install the [DEV version of RF24Network](https://github.com/TMRh20/RF24Network/archive/Development.zip) on the Arduinos.

One option to try with RF24Network is to open RF24Network_config.h in a text editor and comment out the line ``` #define ENABLE_DYNAMIC_PAYLOADS ``` if you suspect that your modules do not support dynamic payloads. The main issue here, of course, is that your payload sizes need to be predetermined for different payload types etc.

Unless you have an LNA + PA model of rf24 module, power supply shouldn't be a problem on RPi.

TMRh20, 2015-08-31:
FYI, if this was the problem, I've finally merged Dev into master, because it seemed the disparity was confusing a lot of people, and it's not always the easiest thing to identify.

closed 2015-09-24 by TMRh20