nRF24/RF24Network #56: Less packet loss when touching module

trojanc, 2015-09-01:
Hi

Firstup, thankyou for a fantastic library!
I have a question regarding packet loss. I wrote a simple test program that sends and receives a small packets at a fixed rate (100ms frequency). The two nodes do exactly the same. 

When my timeout is that short (100ms) I get a lot of packet loss in both receiving and sending on both nodes (about 80%+ gets lost), BUT as soon as I touch one of the RF24 Modules, the traffic sends from the one node at almost no packet loss, and the same happens if a touch the other node's RF24 module

When using a longer timeout (500ms) I got almost no packet loss (without touching the module).

Do you have any idea why touching the module helps with packets loss? And why do I get almost no packet loss by just increasing my frequency too 500ms. I've tried different modules (the long range with antennas, and the cheaper ones) they all behave the same. Both modules are on my table.

I will upload my testing code this evening, and maybe put a video on youtube if I get the chance.

If there is an other forum I should rather ask this question please guide me :) I figured you'd know best.

TMRh20, 2015-09-01:
Np, thanks for the detailed description of the problem.

My first suggestion would be to try setting ```  radio.setPALevel(RF24_PA_MIN); ``` immediately after ``` network.begin(/*channel*/ 90, /*node address*/ this_node); ``` is called.

With 2 nodes in close proximity, they can kind of overpower each other, and touching them seems to reduce this effect in many cases. 

Beyond that I would prob. need to see your code, but I have encountered the exact same issue with PA level set to max.

trojanc, 2015-09-01:
Hi, thank you for your response.

Lowering the PA level did seem to improve it.
Here is a link to my code: https://github.com/trojanc/RF24-Test

It build with platformio - just because I find it simpler to work with than the Arduino IDE. 

This probably isn't the best code to check for packet loss...This was my first attempt. What I eventually want is a program that I can use to test the range of different modules in different situations. For example I need to install monitors inside and outside of a building, and I would like the program to give me an idea of how successful the communication would be, from which I can figure out if I maybe need to add another node in between to help "extend" the range to the nodes outside of the building. 

...Maybe package it all nicely in a plastic box with a LCD screen on both sides so I can instantly see what is going on... and have options for different types of tests to perform (packet loss, throughput, different PA levels, etc...)
That's the bigger picture... for now I just want to figure out what is the best way to check for packet loss. And this thing where it all of sudden start working when I touch the module almost had me loose faith in the modules (and almost made me loose my mind)

 I'm not too worried about transfer speeds, the monitoring data I send is small enough to fit into a 16 byte packet (maybe one day when things get more busy I'll worry about better bandwidth). The data is small but it is critical, so packet loss should be minimal. But I also realize that it's also up to my program to make sure itself handles retries properly ;)

TMRh20, 2015-09-01:
Interesting. With the network library, you could extend the functionality a bit more, but it would require some changes to the code. I've been playing around with it, and initially included similar functionality in RF24Mesh, to determine the quality of the signal level when trying to find a parent node to 'attach' to.

The RF24 core lib includes a function [radio.testRPD();](http://tmrh20.github.io/RF24/classRF24.html#a821285f3b54553f4402eb3fd0ac6d6c1) which can indicate whether the last received payload was 'heard' at greater or less than 64dB.

This provides you with a 3-state signal quality reading, if you are sending packets at timed intervals, and say, expecting a payload every 1/2 seconds.
a: Signal > 64dB
b: Signal < 64dB
c: No msg received in timeout period (No signal)

If you want to push it further, you can also use multicasting when the signals start to degrade, to determine if only one node or both are having trouble writing to the other node. (auto-ack requires two-way comms for success, multicasting is one-way)

This would require some changes to RF24Network to make testRPD readings happen, but they are relatively minor from what I remember.

trojanc, 2015-09-01:
Cool, thanks for all the info. I'll close this "issue"...it was more meant as a chat than an issue really :)

closed 2015-09-01 by trojanc