nRF24/RF24Network #61: Creating a new release

Avamander, 2015-09-26:
Could you please create a new release so that this library can be submitted to the IDE's library repo?

TMRh20, 2015-09-26:
Heh, I assumed you were going to be doing RF24Network also, so I did [RF24Network](https://github.com/TMRh20/RF24Network/releases) and [RF24](https://github.com/TMRh20/RF24/releases)  :p

Avamander, 2015-09-28:
@TMRh20 You have to retag the RF24 library release. The 1.0.0 release has no library.properties tag.

TMRh20, 2015-09-28:
Not sure I did it entirely correctly, but it should be updated with the properties file. Tagged as v1.01.

TMRh20, 2015-09-29:
You now have collaborator access to RF24, RF24Network and RF24Mesh. I hope this makes things easier for you, but please do not push any large changes to the branches without discussion. Thanks!

closed 2015-09-29 by Avamander