nRF24/RF24Network #63: RF24Network and Gateway

spiri439, 2015-10-05:
Hello,
I have the current setup, the gateway is setup with ncurses on rpi2 module and the http server is running there too. I have 5 arduino pro mini with ip 10.0.0.11- to 16 and after 12-24 hours they stop transmitting the data they are supposed to via get(temperature, humidity and etc), if i try to ping them they respond to the ping the mesh sees them, is this a bug or a problem in my configuration , i am supplying current from the pro mini directly, not enough current, or millis() crush ?
Thank you

Avamander, 2015-10-05:
Unfortunately this is a bit wrong place to post this bug report. (I'd suggest you'd check your code if it can handle millis starting from 0)

closed 2015-10-05 by Avamander

TMRh20, 2015-10-05:
If they are responding to ping requests, it might be a bug in your code or possibly RF24Ethernet. If, for some reason, the client thinks it is still connected from the last request, it would no longer attempt to connect.

You could possibly attempt to work around the issue by calling client.stop(); if the client is still connected after a set period of time or something like that. Most of my testing and usage right now is with Arduino as a server, but it seems very stable now, so I will likely be doing more client side testing as time, etc permits.

If no luck, per above, please open an issue under [RF24Ethernet](https://github.com/TMRh20/RF24Ethernet/issues) with your code, or a working example that demonstrates the issue.