nRF24/RF24Network #64: Type (for message header) changing after write?

Skorium, 2015-10-21:
Hi,

I'm completely newbie with the module, I found your libs, blog and github I will say easily. Getting stuffs all arround the web I managed to find [this tutorial] (http://www.homeautomationforgeeks.com/rf24software.shtml) close enough to my needs. I decided to take it as starting point.

Before taking this tutorial, I managed to get different data from different sensors with the same Atmega32 (working with the chip itselft and using Arduino as ISP). Getting data from light sensor, bmp085 for the pressure and temperaure, dht22 for humidity and temperature...

Of course the goal now is to send all these data to my raspberry pi.

Here is my message structure (exactly identical in my receiver and emitter)
```
struct message_t {
  uint16_t als_lux;
  int16_t wdir;
  float wspeed;
  float dht_humidity;
  float dht_temperatureC;
  float dht_heatIndexC;
  float dht_dewpoint;
  float bmp_temperatureC;
  int32_t bmp_atmPressure;
};
```

I have declared the type in both receiver and emitter:
```
header.type = 108;
```

When the Arduino run the setup(), I'm initialising the header. When it goes to the loop() header.type still have 108 as value. Once the write is completed, the value change to 150. So the message is sent every time but treated only the first time by the receiver as the type doesn't match any more.

After looking on Google results a long time, I found something could be related is "#define NETWORK_LAST_FRAGMENT 150" from [RF24Network.h (http://tmrh20.github.io/RF24Network_Dev/RF24Network_8h_source.html) but I have no clue why, and I don't know if this is normal or not. I cannot find related data to this or the research stuff are not good enough to help to find the answer (or I don't know how to make my search, but this is another problem).

As far as I know, I'm not changing this value.
My message should be short enough to be sent at once, am I wrong?
Any idea what can cause this issue? (I believe I'm on the dev branch of RF24Network as the tutorial I followed advice to work with the dev branch).

TMRh20, 2015-10-22:
Thanks for the detailed description and code examples etc. 

I'm just guessing that the problem is that you may be initializing the header only once, instead of creating a new header before every write, which is technically what should be happening.

This ensures the header ID is incremented each time, and things are adjusted accordingly. In this case, your message 32 bytes long and the header is 8 bytes long, so you are making use of fragmentation, and the message type is (probably) not being updated after the first send - because the header is not re-initialized.

Skorium, 2015-10-22:
Thank you for this very quick reply. So because of the fragmentation the value of header type is changing and this is causing the issue? When I figured out the value was changing my 1st action was to reset the value of the header in the loop. And evevery looks working ok.

closed 2015-10-22 by Skorium