nRF24/RF24Network #67: Attiny i RF24Network

moriusz, 2015-11-20:
Hello i have problem whith RF24Net and Tinny devices.. as much RF24 works correct on tinny, network not ... 
I check both cores:
https://github.com/damellis/attiny
http://drazzy.com/
Both works correct whit RF24 but still no comunication over net...
Any clue or sugestions?
Cheers
Peter

TMRh20, 2015-11-21:
If I remember correctly, the ATTiny code was originally tested with [this](https://code.google.com/p/arduino-tiny/) core. I remember commenting about looking at updating this, but never got around to it.

There are also a few differences between RF24Net on attiny, being that fragmentation is disabled, and it uses a slightly smaller buffer, but it should work otherwise. I need to do some testing with ATTiny soon, so I can verify functionality also.

moriusz, 2015-11-21:
I will update my attiny using that core and check if net work as i told both cores work correct whit RF24.. main recivers are on Uno/mini pro compatibile ... i send only reading from DTH when value is change.. on Uno compatibile is working on tinny not and i wstill have free sram and eeprom .. that is not a problem whit to much code putet in i give know leter on about test on arduino tinny core..

moriusz, 2015-11-22:
Tests done still no Net transfer ... RF24 working but net no this is no core issue....

TMRh20, 2015-11-25:
Thanks for the input and testing! 

It turns out, I neglected to enable dynamic payloads for ATtiny after making it an optional feature that is enabled by default. You can test this out/verify by updating to the latest GIT version https://github.com/TMRh20/RF24Network/archive/master.zip or by adding ``` #define ENABLE_DYNAMIC_PAYLOADS ``` to RF24Network_config.h in the ATtiny section lower down.

TMRh20, 2015-11-25:
I forgot to mention, I tested it with the https://github.com/SpenceKonde/ATTinyCore from the link you provided, since this seems to be the most up-to-date and feature rich one available. It works fine using the internal SPI logic per the RF24 ATtiny examples, and may work using the included SPI lib, but is untested.

moriusz, 2015-11-26:
OK  i done the quick test on 85 still no transfer .. when using tc rx example from net repo. Tommorow i spend more time and once again try to make a refer connection o refer core and try once again whit 84 and 85 on tx rx and i post results. At this moment is still this same RF24 simple node transfer woks well net tranfer don't work ...  but as i told i need a some time to check once again whit refer core and  SpenceKonde core

closed 2015-12-26 by Avamander