nRF24/RF24Network #69: New Revision compatable with previous releases?

robwestbrook, 2015-11-25:
During the summer I built a small RF24 network using a RasPi and two arduino pro minis. One pro mini has a DHT22 temp and humidity device and a barometric pressure device connected. It relays its data through the other pro mini, acting as a network extension node, to the RasPi. Both the RasPi and pro minis have the NRF24l01+ PA with external antennae. This setup works superbly. This week I decided to add a couple of new nodes. The Arduino IDE informed me of library updates for RF24 and RF24Network. I updated. Using previously good code no longer works. My existing network never receives any data from the new nodes. Are there compatibility issues I'm not aware of? Is there any way I can get the previous RF24 and RF24Network releases?

TMRh20, 2015-11-25:
See the https://github.com/TMRh20/RF24Network/tree/Archived-OriginalUpdated branch

There should be no compatibility issues with RF24, just the updated network lib.

robwestbrook, 2015-11-25:
Got the archived RF24Network. Can't establish comm with existing RF24Network. So took a couple of nodes and used the new RF24 and this archived RF24Network to network them. New network works great. Now, I've got two networks sitting side by side on bench. Both on channel 90. Both have 00 node and 01 node. Both networks running great. But neither talking to one another. I must be missing something. Got any ideas?

TMRh20, 2015-11-25:
My best guess is that you must have grabbed a previous DEV version of RF24Network. There have been some major changes since the summer to accommodate clone modules, which require updating all nodes, so you could try https://github.com/TMRh20/RF24Network/tree/546e238897d0a0a5bd7b2c2b93ef48b3a0584845 which was Aug 7 before those changes.

robwestbrook, 2015-11-25:
I grabbed the above library. Looks like I have some type of addressing problem. Here's the print details from both the RasPi, which I'm trying to connect to, and the Pro Mini, the node I'm trying to connect.

Here's the details from the RasPi.

    STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
    RX_ADDR_P0-1	 = 0xf0f0f0f03c 0xf0f0f0f05a
    RX_ADDR_P2-5	 = 0x69 0x96 0xa5 0xc3
    TX_ADDR		 = 0xe7e7e7e7e7
    RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
    EN_AA		 = 0x3f
    EN_RXADDR	 = 0x3f
    RF_CH		 = 0x4c
    RF_SETUP	 = 0x27
    CONFIG		 = 0x0e
    DYNPD/FEATURE	 = 0x00 0x00
    Data Rate	 = 250KBPS
    Model		 = nRF24L01+
    CRC Length	 = 16 bits
    PA Power	 = PA_MAX

Here's the details from the Pro Mini

    STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
    RX_ADDR_P0-1	 = 0xf0f0f0d23c 0xf0f0f0d25a
    RX_ADDR_P2-5	 = 0x69 0x96 0xa5 0xc3
    TX_ADDR		 = 0xf0f0f0f069
    RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
    EN_AA		 = 0x3e
    EN_RXADDR	 = 0x3f
    RF_CH		 = 0x4c
    RF_SETUP	 = 0x25
    CONFIG		 = 0x0e
    DYNPD/FEATURE	 = 0x00 0x00
    Data Rate	 = 250KBPS
    Model		 = nRF24L01+
    CRC Length	 = 16 bits
    PA Power	 = PA_HIGH

Can you make anything out of this? There are differences in the RX_ADDR_P0-1, TX_ADDR, EN_AA, and RF_SETUP.

robwestbrook, 2015-11-25:
Here's the code on the Pro Mini. It's your helloworld_tx, with changes from my wiring.

    #include <RF24Network.h>
    #include <RF24.h>
    #include <SPI.h>
    #include <printf.h>
    
    RF24 radio(9,10);                    // nRF24L01(+) radio attached using Getting Started board 
    
    RF24Network network(radio);          // Network uses that radio
    
    const uint16_t this_node = 02;        // Address of our node in Octal format
    const uint16_t other_node = 00;       // Address of the other node in Octal format
    
    const unsigned long interval = 5000; //ms  // How often to send 'hello world to the other unit
    
    unsigned long last_sent;             // When did we last send?
    unsigned long packets_sent;          // How many have we sent already
    
    
    struct payload_t {                  // Structure of our payload
      unsigned long ms;
      unsigned long counter;
    };
    
    void setup(void)
    {
      Serial.begin(9600);
      Serial.println("RF24Network/examples/helloworld_tx/");
      SPI.begin();
      printf_begin();
      radio.begin();
      radio.setDataRate(RF24_250KBPS);
      radio.printDetails();
      network.begin(/*channel*/ 90, /*node address*/ this_node);
    }
    
    void loop() {
      
      network.update();                          // Check the network regularly
    
      
      unsigned long now = millis();              // If it's time to send a message, send it!
      if ( now - last_sent >= interval  )
      {
        last_sent = now;
    
        Serial.print("Sending...");
        payload_t payload = { millis(), packets_sent++ };
        RF24NetworkHeader header(/*to node*/ other_node,1);
        bool ok = network.write(header,&payload,sizeof(payload));
        if (ok)
          Serial.println("ok.");
        else
          Serial.println("failed.");
      }
    }

And the strange output I'm getting:

    Sending. . . ok.
    Sending. . . ok.
    Sending. . . ok.
    Sending. . . ok.

There's no response on the RasPi end.

TMRh20, 2015-11-26:
So the ``` 0xf0f0f0f03c ``` addressing tells me for sure you were using the https://github.com/TMRh20/RF24Network/tree/Archived-OriginalUpdated branch. 

The EN_AA        = 0x3e tells me that you may not have power cycled the Arduino/radio after changing between libraries, because pipe0 has AA disabled.

My best suggestion would be to remove any and all previous versions of RF24Network and download [this](https://github.com/TMRh20/RF24Network/archive/Archived-OriginalUpdated.zip) version. 

Extract it to your libraries folder and rename to RF24Network.

Then start/restart the Arduino IDE and re-upload to Arduino. Once complete, unplug Arduino from all power sources, and power up. Everything should work.

robwestbrook, 2015-11-26:
Cycling power worked...with the RF24Network-546e238897d0a0a5bd7b2c2b93ef48b3a0584845 branch. Now I'm filling up the RasPi buffer with the same packet. I have my RasPi displaying node ID, header type, and Packet ID with a network.peek on every received packet. I now have the Pro Mini powered off but my RasPi is spitting out packet info as fast as it can. Running my RasPi side with python. Is there someway to clear the buffer after every packet?

robwestbrook, 2015-11-26:
OK, I see that a network.read clears the buffer. Everything is a go now. Thank you so much for your help!
Closing.

closed 2015-11-26 by robwestbrook