nRF24/RF24Network #73: Problem in communication between arduino and rpi

wilmsn, 2015-12-28:
I got a problem in communication between arduino uno and rpi typ 2. Sending from arduino to rpi is OK but receiving not. RPI is Node "00" and Arduino is Node "02". When i go in debug mode i see(diffs are marked **bold**):
RPI:
    ================ NRF Configuration ================
    STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
    RX_ADDR_P0-1     = 0xcccccc**cc**c3 0xcccccc**cc**3c
    RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
    TX_ADDR          = 0xcccccc33e3
    ----
    943321: MAC Sending to 02 via 02 on pipe 5
Arduino:
    STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
    RX_ADDR_P0-1	 = 0xcccccc**3c**cc 0xcccccc**33**3c
    RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
    TX_ADDR		 = 0xcccccccc33
    ----
    6265: MAC Sending to 00 via 00 on pipe 2
TX-Address in Arduino matches with RX-Address in RPI (pipe 2) and communication ist OK.
RX_ADDR_P0-1 on Arduino are different in marked diggits. Is that normal or a bug in SW?
I tried some nrf24l01+ modules and the result is always the same.

TMRh20, 2015-12-28:
The addressing looks correct, all nodes use unique RX addresses.

Chances are your CE pin is not set correctly on RPi if you cannot transmit, but you would usually get an error message. See bottom of page [here](http://tmrh20.github.io/RF24/RPi.html) for default pins.

A simple method when coding is to use a minimal constructor like ``` RF24 radio(22,0); ``` where 22 is any GPIO pin number not designated for SPI, and 0 is the SPI BUS number (labeled CE0 or CE1 but used for radio CSN)

See http://www.megaleecher.net/sites/default/files/images/raspberry-pi-rev2-gpio-pinout.jpg

I would suggest testing with the core RF24 examples if no luck with RF24Network, just to make sure the hardware & config is all OK.

wilmsn, 2015-12-30:
Thanks for response! It seems to be a problem with the hardware. With other hardware everythings runs fine and smooth. Good work - thanks!

closed 2015-12-30 by wilmsn