nRF24/RF24Network #74: Warings while compiling

JorgMuskens, 2016-01-06:
Hi,
I got the following warnings while compiling

C:\Program Files (x86)\Arduino\libraries\RF24\RF24.cpp:569:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

   printf_P(PSTR("Data Rate\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_datarate_e_str_P[getDataRate()]));

C:\Program Files (x86)\Arduino\libraries\RF24\RF24.cpp:570:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

   printf_P(PSTR("Model\t\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_model_e_str_P[isPVariant()]));

C:\Program Files (x86)\Arduino\libraries\RF24\RF24.cpp:571:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

   printf_P(PSTR("CRC Length\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_crclength_e_str_P[getCRCLength()]));

C:\Program Files (x86)\Arduino\libraries\RF24\RF24.cpp:572:17: warning: invalid suffix on literal; C++11 requires a space between literal and identifier [-Wliteral-suffix]

   printf_P(PSTR("PA Power\t = "PRIPSTR"\r\n"),pgm_read_word(&rf24_pa_dbm_e_str_P[getPALevel()]));

The pointer on all 4 warnings points to the first " in the line of code. I tried adding the space between the literal and identifier but that did not work. Any ideas?

Regards Jorg

TMRh20, 2016-01-06:
They are just warnings, so it should still work ok. I'll take a look at clearing the warnings as soon as I get a chance.

Skorium, 2016-02-27:
Hi guys, I got the same warnings there.

Very easy to fix, should be fixed in less than 30sec, re-compilation of your project included ;)

In RF24.cpp go to line 569, 570, 571, 572 and add a blank space before each PRIPSTR as below:

`  printf_P(PSTR("Data Rate\t = " PRIPSTR"\r\n"),pgm_read_word(&rf24_datarate_e_str_P[getDataRate()]));
  printf_P(PSTR("Model\t\t = " PRIPSTR"\r\n"),pgm_read_word(&rf24_model_e_str_P[isPVariant()]));
  printf_P(PSTR("CRC Length\t = " PRIPSTR"\r\n"),pgm_read_word(&rf24_crclength_e_str_P[getCRCLength()]));
  printf_P(PSTR("PA Power\t = " PRIPSTR"\r\n"),  pgm_read_word(&rf24_pa_dbm_e_str_P[getPALevel()]));`

Save changes, re-compile your project (I didn't have to restart my Arduino IDE).

Enjoy your compilation warning free :)

TMRh20, 2016-03-23:
This should be fixed now

closed 2016-03-23 by TMRh20