nRF24/RF24Network #77: Address "05" fail

DavidNovoa, 2016-02-01:
Hi,
I disable multicast but the address 05 not work for me. Using the helloworld_tx and helloword_rx if the address of TX is 05 not work. Helloword_rx received the message but helloworld_tx displays "Sending...failed".

Thanks

TMRh20, 2016-02-20:
I think the disabling of multicast may be broken for the time being, due to recent changes to how the multicast pipe/address is handled. 

Will update this once resolved.

DavidNovoa, 2016-02-22:
I got to work looking at the history and rescuing some pieces. In my case it works, but do not fully understand the program and do not know if it is correct.

You can see it in my fork.
[https://github.com/DavidNovoa/RF24Network/commit/1004f7a28278d8abb600ea9240ad1ca0b53b7903](url)

TMRh20, 2016-02-23:
Nice attempt. It is a complicated issue to fix, and I think there are still some potential problems, but this is a start. Hopefully I'll get a chance to look closer at this and test things out soon.

TMRh20, 2016-03-23:
I found an issue with node 05 pipe assignment when sending per https://github.com/TMRh20/RF24Network/commit/3a42eb9d8e4764f1c007fd4ee2517c13e37b764e

With multicast enabled, the master node can have 5 children, but all other nodes can only have 4, so I think disabling of multicast might still be broken, but this particular issue might be fixed.

izero79, 2017-07-06:
Hi,

I wonder if I have this same problem or a different. I'm using RF24Network (v 1.0.8) on Arduino nodes and py24Network as master with RPi.

I currently have network with master 00 and 7 nodes 01-04 + 013-033 and I'm adding a new node with address 05. However I always get send failed for the address 05. The communication actually works ok between 00<->05 but the 05 node always gets 'false' for the network.write() so I cannot find out in code that was the message actually sent ok or not.

If I change the node address to anything else like 01-04 or 043 the network.write returns 'true'.

I tried to set the multicast to disabled/enabled but doesn't seem to have any effect to this.

Any ideas?

closed 2019-05-10 by TMRh20