nRF24/RF24Network #78: Alignment issue on ARM

ab316, 2016-02-05:
I have ported RF24 to ARM Cortex M0+ STM32L053.
The enqueue method of RF24Network class fails when the frame_buffer is accessed through the pointer as in:
	memcpy(next_frame,&frame_buffer,8);
	RF24NetworkFrame *f = (RF24NetworkFrame*)next_frame;
	f->message_size = message_size;

This is due to unaligned memory access of the message_size field. The CPU goes into hard fault when the field is accessed.
To prevent this, the frame_buffer should be padded so that next_frame only points to addresses that are multiple of 4 (since ARM Cortex M0 is 32-bit) and the padding size must also be stored.

TMRh20, 2016-02-20:
I'll have to look at this and #80 a bit closer but I think you are correct.

ab316, 2016-02-20:
I was able to partially fix it by adding the following lines after
uint8_t frame_queue[MAIN_BUFFER_SIZE]
and
uint8_t frame_buffer[MAX_FRAME_SIZE]

	#if defined(__arm__)
	__attribute__((aligned(4))) /* Need to be memory aligned on ARM. Otherwise will cause hard fault when accessed through a pointer */
	#endif
	;

If you want I can make a pull request for this new header file.

TMRh20, 2016-02-20:
Sure, looks good to me. I only wonder if the #if define is needed? Maybe its better to define the memory alignment all around and work with that?

ab316, 2016-02-20:
Yes, sure.
I was uncertain so I just put it to be on the safe side.
Should I make a pull request with or without the #define.

ab316, 2016-02-20:
The problem is, when the Network receives any arbitrary length message which is not a multiple of 4, the alignment problem occurs again.
What is needed here, in my view, is to add padding in the buffer to ensure alignment. But, then, this padding needs to be recorded and then removed during read operation.
I was thinking of doing that. But due to urgency, I disabled dynamic payloads instead.

Bitobux, 2016-02-20:
I agree, the message length must be a multiple of 4, but the padding in structs must also be solved.
see my issue #80 .

Myidea:
You can disable padding in struscts in this way:
struct __attribute__((__packed__)) RF24NetworkHeader
struct __attribute__((__packed__)) RF24NetworkFrame

After that, the code must be changed in some places in RF24network.cpp:
example: RF24Network::read

the pice of code:
	memcpy(&header,frame_queue,8);
	RF24NetworkFrame *f = (RF24NetworkFrame*)frame_queue;
	bufsize = f->message_size;

can be substituded with:
	memcpy(&header,frame_queue,8);
	memcpy(&bufsize,(frame_queue+8),2);

**Doing this kind of changes I solved issue #80 **
I'm sorry my english is terrible!!!!

TMRh20, 2016-02-21:
Thanks for the suggestions. I started to test this more on the ESP and was encountering the issues myself. I should get a chance to look at incorporating these changes and testing more soon.

TMRh20, 2016-02-21:
Ok, I had a chance to add a couple things tonight. The changes I added to [dev](https://github.com/TMRh20/RF24Network/archive/Development.zip) seem to work OK on ESP with payloads of 8 or 9 bytes, but I haven't had a chance to fully understand the issue, so feel free to add further pull request or info for testing etc.

1. Change message_size to 16-bit in enqueue()
2. Use memcpy instead of frame pointer
3. Add padding to next frame pointer to ensure divisible by 4.

Bitobux, 2016-02-21:
Thanks, I'll test asap.

Bitobux, 2016-02-21:
**IT WORKS!! THANKS!!!**

I checked your modifications:  you decided to don’t pack the RF24NetworkHeader and RF24NetworkFrame structures….  hmmm…. It works because you are lucky, sizeof(RF24NetworkHeader) is 8bytes in both my CPUs (ATmega328 8bits, ESP8266 32bits). With different results of sizeof(RF24NetworkHeader) you also have to modify the code everywhere there is a sizeof(RF24NetworkHeader).

In the documentation: if must be transmitted structures as payload (very probable), is better to warn the user about to verify if sizeof(payload) is equal in both the cpu nodes (to check if the structure padding is the same). In my opinion, if the result is different, the following tx-rx can’t work.
network.write(header,&payload,sizeof(payload)) 
network.read(header,&payload,sizeof(payload))

Bitobux, 2016-02-22:
In my opinion we don't neen the pice of code:    
if(uint8_t padding = (message_size+10)%4){
      next_frame += 4 - padding;
    }
There are troubles only if we try to address a 32-bit word that is not aligned to a 4-byte boundary, using memcpy the problem is already solved.

Bitobux, 2016-02-22:
Also **RF24Network::peek must be fixed**. please substitute **return frame->message_size** with a memcpy

	RF24NetworkFrame *frame = (RF24NetworkFrame*)(frame_queue);
	memcpy(&header,&frame->header,sizeof(RF24NetworkHeader));
	return frame->message_size;   <<<<<<<<ERROR

TMRh20, 2016-02-23:
Thanks!

From my limited understanding, it seems that most devices will pack the data together in the most efficient manner, as long as given a chance. For example, a structure containing uint32_t, uint16_t uint8_t uint8_t seems to usually pack together nicely using 8 bytes. If it was arranged differently, as uint16_t uint32_t uint8_t uint8_t then it will be packed differently. and usually use 12 bytes on a 32-bit system. I guess the question I have is do we have any examples of systems that show the header as other than 8 bytes?

The new code looks good, so I'll update the peek function soon!

closed 2016-03-23 by TMRh20