nRF24/RF24Network #79: Number of "parent" / Main Nodes

marc-gist, 2016-02-07:
I want/need to have more than the 5 nodes communicating with the master/base node... can this be done easily modifying the code?

I have looked at just using the RF24 code to do this.. however, it doesn't have the great features of the RF24 node, like large packet/data chunking, "guaranteed" delivery, etc.

Thanks,

Avamander, 2016-02-07:
If you mean 5 nodes directly connected to the master then, no, you can't,
it has been discussed before https://github.com/TMRh20/RF24Mesh/issues/30.
But indirectly (through another node) you can have hundreds of them.

About ACK functions, as hardware cannot be relied on you must implement one
yourself. As far as large payloads are concerned, RF24Mesh can handle them.

marc-gist, 2016-02-07:
That link doesn't "discuss" much... I get there are a limited number of pipes; but why can another node have hundreds? From the discussion/pipe reasons, I assume that each node level can only have 5 nodes per tree level... so you need to "waste" a node to be a relay for others.  I get this works great for chaining nodes far away from the master; but I want many near the master.
I'll look into getting my code/design to work with relays/chained nodes.
Otherwise I guess my only option is to use the standard RF24 library and implement my own Master/Slave situation.

Thanks!  This is a great library!!!

TMRh20, 2016-02-20:
Relay nodes are not 'wasted' as they can still operate normally. 

The radio limit is manufacturer specified. In order to have more than 6 nodes in communication with a single node, AA/ESB generally needs to be disabled and potentially handled in software.

closed 2016-02-20 by TMRh20