nRF24/RF24Network #80: Can't RX on ESP8266

Bitobux, 2016-02-18:
Thank you TMRh20  for your great work on RF24 and RF24Network libs.

My config:
node 0: ESP8266 + nRF24L01+ mini
node 1: Arduino Pro Mini + nRF24L01+ mini

Wires are OK because:
works with MySensors libraries.
helloworld_tx on ESP and helloworld_rx on Pro Mini works.

ISSUE:
helloworld_rx on ESP  and helloworld_tx on Pro Mini:
Every time node 0 (ESP) receives a message restarts:

Exception (9):
....
rst cause:2, boot mode:(3,6)

Ideas?

Avamander, 2016-02-18:
Please post the full exception.

Bitobux, 2016-02-19:
**On the TX side (Pro Mini):**
Sending...ok.
Sending...ok.
:
**On the RX side (ESP8266):**
RF24Network/examples/helloworld_rx/

Exception (9):
epc1=0x40201b02 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee961 depc=0x00000000

ctx: cont 
sp: 3ffef9d0 end: 3ffefc00 offset: 01a0

>>>stack>>>
3ffefb70:  000000ff 00000010 3ffeeaa0 3ffeebcc  
3ffefb80:  000000ff 3ffee922 3ffeeaa0 00000000  
3ffefb90:  000000ff 3ffee922 3ffee918 40202009  
3ffefba0:  cccccc01 00000000 3ffee918 40201d38  
3ffefbb0:  3ffeeaa0 3ffeeaa0 3ffeeac0 3ffeebcc  
3ffefbc0:  3fffdc20 3ffee918 3ffeebc4 402018b4  
3ffefbd0:  40203559 00000000 3ffeeba0 4020188c  
3ffefbe0:  00000000 00000000 3ffeebc4 40203581  
3ffefbf0:  00000000 00000000 3ffeebe0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

**If i swap the scripts:
On Pro Mini:**
Received packet #36 at 74000
Received packet #37 at 76000
Received packet #38 at 78000

**On ESP8266:**
Sending...ok.
Sending...ok.
Sending...ok.

Bitobux, 2016-02-19:
After some investigations..... the issue is at function:
uint8_t **RF24Network::enqueue**(RF24NetworkHeader* header)
......
  if(message_size + (next_frame-frame_queue) <= MAIN_BUFFER_SIZE){
	memcpy(next_frame,&frame_buffer,8);
	RF24NetworkFrame *f = (RF24NetworkFrame*)next_frame;
	**f->message_size = message_size;         <<<<< BOOOM!!!**
	memcpy(next_frame+10,frame_buffer+sizeof(RF24NetworkHeader),message_size);

Note: there is a similar open issue: **Alignment issue on ARM** #78

Bitobux, 2016-02-19:
The problem is padding.

the struct **RF24NetworkFrame** contains:
8 bytes: RF24NetworkHeader > 8 bites
2 bytes: uint16_t message_size >2 bites
2 bytes: uint8_t *message_buffer 
total: 14 bytes

Serial.print("sizeof(RF24NetworkFrame)="); Serial.println(**sizeof(RF24NetworkFrame)**);
prints **16**

I'm using Arduino IDE.

Bitobux, 2016-02-20:
With some changes my ESP8266  can RX...
please see my comment at issue #78

TMRh20, 2016-03-23:
Fixed

closed 2016-03-23 by TMRh20

jdpanchal, 2019-03-22:
it is possible to use RF24Network.h in esp8266.it s working properly in ardiuno nano. 
error is as below. please give some suggestion, thanks you.

Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

WARNING: library RF24Network-master claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
C:\Users\jigo\Desktop\NRF +ESP\esp8266\esp8266.ino: In function 'void loop()':

esp8266:168: error: invalid conversion from 'const void*' to 'void*' [-fpermissive]

     network.read(header, &incomingData, sizeof(incomingData)); // Read the incoming data

                                                             ^

In file included from C:\Users\jigo\Desktop\NRF +ESP\esp8266\esp8266.ino:14:0:

C:\Users\jigo\Documents\Arduino\libraries\RF24Network-master/RF24Network.h:467:12: error:   initializing argument 2 of 'uint16_t RF24Network::read(RF24NetworkHeader&, void*, uint16_t)' [-fpermissive]

   uint16_t read(RF24NetworkHeader& header, void* message, uint16_t maxlen);

            ^

exit status 1
invalid conversion from 'const void*' to 'void*' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Avamander, 2019-03-22:
@jdpanchal It's an error in your code.