nRF24/RF24Network #83: How to set payload type in pyRF24Network

izero79, 2016-03-07:
This might be something that happens because I'm not very experienced with Python, but I'm not able to set payload type to RF24NetworkHeader correctly from Raspberry using pyRF24Network.

Here's the essential part of my code:

```
                    payload = pack('<BB', alarm_level, sleep_cycles)
                    other_node = octlit("01")
                    print('Sending...')
                    ok = network.write(RF24NetworkHeader(other_node, ord('C')), payload)
                    if ok:
                        print('ok.')
                    else:
                        print('failed.')
```

I get 'ok' print to command line, but other node (Arduino) never receives nothing. However, if I just leave out the payload type ( `network.write(RF24NetworkHeader(other_node), payload)` ) other node receives the payload (with type '0'). I tried to replace the `ord('C')` with 0, but it fails also.

Other node receives the 'C' typed payload correctly if I use C++ version of the RF24Network  on the raspberry or another Arduino. So I'm assuming that I'm setting the payload type in wrong way in Python, but I don't know how I should give it and I couldn't find example for it.

mz-fuzzy, 2016-03-08:
Your python code you posted seems ok. I tried to replicate behavior you described between python RPi and an arduino, but everything is working with me, both in python2 and python3. Maybe you can try to modify helloworld_tx.py by just adding the packet type, run unmodifier helloworld_rx.ino on arduino and see whether it's delivered. If so, then the problem is probably elsewhere in your code.

izero79, 2016-03-10:
I got this working by setting the payload type to variable before creating the header, like this:
```
other_node = octlit("01")
type = ord('C')
print('Sending...')
ok = network.write(RF24NetworkHeader(other_node, type), payload)
```

I don't know what is the difference, but fixes my problem so good enough.

mz-fuzzy, 2016-03-30:
https://github.com/TMRh20/RF24/issues/205#issuecomment-203271486

TMRh20, 2016-03-30:
I don't want to say it, but from the description in the other thread:
> random changes that logically doesn't change the code breaks the sending of payloads.

When completely illogical stuff like that is happening, it always seems to be a memory overwrite or memory issue somewhere causing it.

TMRh20, 2016-03-30:
@mz-fuzzy Are you using the latest RF24Network code? Im thinking about the changes to [Fix alignment issues on ARM devices](https://github.com/TMRh20/RF24Network/commit/0e102a24c9dee7335a9fd0d0f7b56a6f3fbb39f1)

izero79, 2016-03-30:
Well I am currently using the latest version of RF24 and RF24Network. But I updated to the latest commits just 2 days ago and before that I was using RF24 Feb 9 commit 47435fdd and RF24Network Dec 23 commit 48125041. Same problem with both of these versions.

TMRh20, 2016-04-01:
@izero79 Thanks! That's pretty much what I wanted to know.

izero79, 2016-04-12:
I made a small pair of example code that I can use to reproduce the problem. Can be found from here: 
[https://github.com/izero79/rf24bug_example]

Simple sender for Arduino and receiver+responser for Raspberry Pi. When I run the RasPi code it fails after few received payloads and sending fails (Arduino doesn't receive anything). Then when I remove the comment block totally the code runs fine in RasPi.

I hope this helps to find out the problem.

Example output, failing code:

```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ce3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x64
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

------------------------------------
2016-04-12 15:25:27
Measurements from: 01
------------------------------------
Message No:  41
o ok.

------------------------------------
2016-04-12 15:25:29
Measurements from: 01
------------------------------------
Message No:  42
o ok.

------------------------------------
2016-04-12 15:25:30
Measurements from: 01
------------------------------------
Message No:  43
o ok.

------------------------------------
2016-04-12 15:25:32
Measurements from: 01
------------------------------------
Message No:  44
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
RF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
^CRF24 HARDWARE FAIL: Radio not responding, verify pin connections, wiring, etc.
Traceback (most recent call last):
  File "./forgithub.py", line 96, in <module>
    send_reply(str(oct(header.from_node)), messageno)
  File "./forgithub.py", line 52, in send_reply
    ok = network.write(RF24NetworkHeader(octlit(node), type), payload)
KeyboardInterrupt
```

The comment block removed:

```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc33e3
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x64
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

------------------------------------
2016-04-12 15:29:33
Measurements from: 01
------------------------------------
Message No:  0
o ok.

------------------------------------
2016-04-12 15:29:34
Measurements from: 01
------------------------------------
Message No:  1
o ok.

------------------------------------
2016-04-12 15:29:36
Measurements from: 01
------------------------------------
Message No:  2
o ok.

------------------------------------
2016-04-12 15:29:37
Measurements from: 01
------------------------------------
Message No:  3
o ok.

------------------------------------
2016-04-12 15:29:39
Measurements from: 01
------------------------------------
Message No:  4
o ok.

------------------------------------
2016-04-12 15:29:40
Measurements from: 01
------------------------------------
Message No:  5
o ok.
```

izero79, 2016-04-15:
I noticed that there was recent commit in RF24: 'avoid memory corruption when calling RF24::read(0) on linux machines' so I updated the lib and it seems to fixed this issue also. My test code and "production" code works now fine.

izero79, 2016-04-15:
... and I just found out that that wasn't the case after all. Doing some more changes to my code triggered the problem again.

mz-fuzzy, 2016-04-16:
@izero79 can you share commit shas of RF24 and RF24Network which give this error on your examples. And also which python version please.

mz-fuzzy, 2016-04-16:
oh, ok, I see that information now earlier in the thread.

mz-fuzzy, 2016-04-16:
I can see this behavior now... In my opinion it points to uninitialized memory read somewhere. I'll try to find it.

mz-fuzzy, 2016-04-16:
@TMRh20 it seems there are multiple unitialized stuff in there like networkFlags or returnSysMsgs? How it is supposed to be set? Not sure whether it has really impact on this issue but I experienced different behavior in test runs due to this.

TMRh20, 2016-04-17:
Most likely they should be null/0 if they are uninitialized. I can take a
closer look and see about correcting that.

mz-fuzzy, 2016-04-18:
@TMRh20 thanks for the fix, I'll check...

mz-fuzzy, 2016-04-18:
@izero79 can you please repeat your test with latest RF24Network change a00090cae05063d17a05763e0bc92bd0ee8762c3 and report back your findings.

I cannot see the erroneous behavior there anymore, but this can be also by changed memory content after the change. If you see the error again, it would be good if you update your error-exposing example, thanks.

izero79, 2016-04-18:
Looks good. I have made many changes to my code and have not got any errors, so it seems to be fixed for me. I will report again if I face this issue again.

mz-fuzzy, 2016-04-18:
closing for now as the bug seems to be fixed.

closed 2016-04-18 by mz-fuzzy