nRF24/RF24Network #84: Network_Ping_Sleep Compile Error

mmackay03, 2016-03-21:
I am trying to upload the Network_Ping_Sleep.ino example to my Arduino Mini Pro, but am getting the following error:

> Network_Ping_Sleep.cpp.o: In function `setup':
> C:\Program Files (x86)\Arduino/Network_Ping_Sleep.ino:104: undefined reference to `RF24Network::setup_watchdog(unsigned char)'
> Network_Ping_Sleep.cpp.o: In function `loop':
> C:\Program Files (x86)\Arduino/Network_Ping_Sleep.ino:139: undefined reference to `RF24Network::sleepNode(unsigned int, int)'
> collect2.exe: error: ld returned 1 exit status
> Error compiling.

I am hoping to use the sleepNode() method in an Arduino sketch of my own. However, I am having a difficult time getting anything to compile that includes sleepNode.
Also, it is worth noting that I uncommented #define ENABLE_SLEEP_MODE in RF24Network_config.h and that the other example files are working correctly. I would appreciate any help that you can offer.

TMRh20, 2016-03-23:
Make sure to remove any old library installations. Otherwise it should work if enabled.

mmackay03, 2016-03-23:
Thanks for the fast reply! I actually just figured it out. It was an issue with me enabling it for ATTiny and not non ATTiny. I have the Network_Ping_Sleep compiled now, but the sleepNode method does not seem to be giving me the power conservation I was hoping for. It seems the radio is not actually sleeping.
I am building a wireless home security system for my senior capstone project using a Raspberry Pi and Arduino Pro Mini. I am trying wanting to power the Arduino with 2AA, however, I am concerned about how long the batteries will last. With Network_Ping_Sleep the Arduino is drawing about 16mA when not transmitting which leads me to believe that the radio does not actually sleep. I currently am implementing powerDown() in my project to conserve the battery. The Arduino is drawing about 2.2mA when not transmitting while powering down both the radio and Arduino. However, this does not let me wake the radio with an interrupt which is why I am hoping to implement sleep mode. The documentation in the RF24Network.h says that in sleep mode the radio will use as little as 22ua. I am wondering if I am missing something and would appreciate any help or advice you could offer.
Thanks in advance for the help and thank you for all your hard work on the RF24 project.

TMRh20, 2016-03-24:
~15mA is the number. RF24Network will always revert to listening mode immediately after sending, so the radio will be actively listening most of the time.

In sleep mode, the radio is effectively 'off', but retains all settings.
Using sleep mode with interrupts, allows the MCU (Arduino) to sleep, but the radio still is actively listening (15mA)

You typically need to design the network for minimal power consumption. Leaf nodes can fully sleep if not needing to receive data, and routing nodes generally need to stay on. Compared to WiFi devices, 15mA is pretty reasonable, but of course a node using full power down and sending data every minute or so can last years off standard batteries compared to days.

See https://github.com/TMRh20/RF24/issues/195 for some more info on radio modes vs power consumption. Needs to be added to docs and examples fixed.

mmackay03, 2016-03-30:
Thank you for the help! That makes sense. I think I will leave it the way I have it now were it is 2.2mA by shutting down the Arduino and the radio.
Sorry, I just have one more question. Is the Sync functionality "completed" or still in development? I have an issue with the Sync::update() when implementing. I noticed the make file does not seem to include the Sync.cpp to compile, however, after compiling it I still have an issue with the update function. I am hoping to use the Sync class to sync the status of my alarm system across all the nodes.
Once again, thank you so much!

TMRh20, 2016-04-01:
The sync stuff was written by @maniacbug originally, and to be honest, I've never used it but have left it in place because people may find it useful. A simple alternative may be to have nodes request the info from the master at set intervals, but it depends on what you are doing I guess.

mmackay03, 2016-04-01:
That's what I figured. I planned on having the nodes request the data if I can't get Sync to work. Thank you so much for all the help! Great work with the library.

closed 2016-04-01 by mmackay03