nRF24/RF24Network #85: Question about multicastRelay and multicasts to child nodes.

harmon25, 2016-04-04:
I love this library, it has motivated me to get more into c++ !
Am working on a project to create a reasonably secure wireless sensor network on the back of RFNetwork. Will probably have more questions in the future... :v: 

When setting mutlicastRelay to true on a node, that node is no longer receiving any multicast packets.

The [documentation](https://tmrh20.github.io/RF24Network/classRF24Network.html#add7408ca2b2f9ab15331ba77a747f7bb) mentions:
`Forwarded payloads will also be received.` 

Is that incorrect or am I missing something? 

My objective is to have heartbeat packets sent via multicast to all nodes. 

At the moment am only testing with a few nodes so a multicast on level 1 is sufficient to make sure they all receive the multicast. 
With more nodes, and thus some children, would the solution be to send a multicast on each level from the master?

Also when creating a header for multicast, what the should the `_to` parameter in the constructor be set to? Been using 00 cause it doesn't seem to matter...

Would it be worth it having a header constructor for crafting multicast headers with just a type parameter? 

Thanks!

harmon25, 2016-04-04:
I have not looked much into the `Sync` module.
Would that be the right approach to keep a heartbeat synchronized on all nodes?
Have noticed latency to be an issue with nodes not directly communicating with the master, presume any `Synced` variable would also suffer from that latency?

TMRh20, 2016-04-04:
In theory it should work both ways, the payload is loaded into the network buffer, then sent if forwarding is enabled per https://github.com/TMRh20/RF24Network/blob/master/RF24Network.cpp#L245  so the only difference with it enabled should be the additional send.

 With multicast the _to parameter doesn't matter, since it is going to everybody. You should be able to use ``` RF24NetworkHeader header; ``` to send with no type. A header constructor without _to would be easy enough to add I think.

The sync module was written by @maniacbug and left in place in case users find it useful. Latency will increase for every 'hop' over a relay node, since the payload is re-transmitted.

mmackay03, 2016-04-04:
@harmon25 Please let us know if you are able to implement the Sync module. I attempted to use it on my project but had issues with the update function in Sync.cpp

harmon25, 2016-04-04:
@TMRh20 I will turn on debugging and give it another shot...

- Just to clarify, **by default**, a multicast to level 1 from the master(00), without multicastRelay enabled should only reach the children of the master (01,02,03,04,05).
- With multicastRelay=true, the children should automatically forward lvl1 multicast packets to lvl 2, which would  reach the grandchildren (011, 012, 013, 014, 015).

- If I wanted to customize the multicast levels, does the parent->child relationship have any bearing? By **default** their seems to be a connection between multicast level, and the nodes logical position on the network.
- If for example i wanted all the nodes in one room to share a multicast level, would they all have to be direct children of the master, or would a configuration such as the below work?

Room A
 - Nodes: 01 (p=00), 03 (p=00), 014 (p=04), 015 (p=05)
 - multicast lvl 1

Room B
 - Nodes: 02 (p=00), 04 (p=00), 025 (p=05), 05 (p=00), 013 (p=03)
 - multicast lvl 2

Not a very efficient topology, just to demonstrate the question...

RF24Mesh does not have a multicast method, is it feasible to multicast too all nodes on an RF24Mesh network?

@mmackay03 Will let you know if I have any success with Sync!

TMRh20, 2016-04-04:
Just think of 'multicast levels' as a shared address with no ACKs.

All nodes listening to a shared address can receive data broadcast to that
address from any other node.

Point 1: Yes

2. Yes. If nodes are close together, you only need 1 multicast-relay per
level/group.

3. The default mulitcast 'levels' are just organized that way because it
seemed logical, and nope there is no bearing between levels and
parent/child.

4. Yes, any configuration. All nodes could share 1 level.

5. Yup, RF24Mesh can be seen at as a simple protocol for addressing in
RF24Network. At its core, all it is, is a standardized process to contact
the master node, & request an address. This means that communication
from/to *node <-**> master *can use RF24Network directly, multicast
included. (master typically performs address translation using its list of
nodes before sending)

harmon25, 2016-04-05:
Thanks!

Tested multicast relay more on my particular setup, Raspberry Pi Master, Arduino Children.

Can confirm the multicast packets are being forwarded, but **not** received as stated in the documentation. 

- pumping multicast messages from 00 on level 1
- testing with 3 nodes (00 => 01 => 011)
- multicastRelay=true set on node 01
-  the child 011 gets the heartbeat, however 01 never receives it, even though it must be forwarding it.

Cannot seem to get SERIAL_DEBUG_ROUTING to work, or any SERIAL_DEBUG within the library for that matter. Any tips on that?  

>  4. Yes, any configuration. All nodes could share 1 level.

All nodes?!
So for my use case where all nodes need to receive the heartbeat, regardless if a node is a child or grandchild they can all be on the same level, with 6+ nodes? That would be great!

If that is the case, wont worry about relaying heartbeats, but have everything on the same multicast level, for now anyway...

TMRh20, 2016-04-05:
Haven't been able to verify, but debug needs printf:
```
#include <printf.h>
```
then
```
Serial.begin(115200);
printf_begin();
```

closed 2019-05-10 by TMRh20