nRF24/RF24Network #87: No interrupts from nrf24l01+ while sleeping  when using radio.stopListening()

aaddame, 2016-04-10:
I'm trying the following setup:
N011 -> N01 -> Master.

N011 and N01 are devduino's running on batteries, so I want them to sleep while not sending data.
I use the sequence radio.stopListening(); network.sleepNode(8,0);

When N011 sends data to the master, N01 doesn't wake up to route this frame. So the radio doesn't force an interrupt in this case.
The code works as expected if I leave out the radio.stopListening();
Attached is the code I use for the nodes. radio.stopListening(); is commented out

[Sensor_DevDuino_Sleep.zip](https://github.com/TMRh20/RF24Network/files/211775/Sensor_DevDuino_Sleep.zip)

TMRh20, 2016-04-10:
Unfortunately the RF24Network documentation and examples need some updating and clarification, but this is expected behavior. 

The radio can only trigger an IRQ if actively listening. You can sleep the MCU, but not the radio if you want it to wake up via interrupt. If sleeping the radio, you can use the WDT or another source like a button etc. for wakeup. 

The only exception to this is when transmitting using ACK payloads, since the radio will automatically enter RX mode to receive the ACK, and an interrupt will be triggered in this case. This is not used in RF24Network.

See https://github.com/TMRh20/RF24/issues/195 also

aaddame, 2016-04-11:
So am I correct saying the Network_Ping_Sleep example doesn't work?
It also uses the stopListening, sleepNode sequence. Every time through the loop it sends a T-frame to one of the known nodes, but chances are this node is sleeping and it will never receive this frame.
The whole neighbor discovery depends on receiving T-frames. Because the node is sleeping most of the time, this is very unlikely to happen.

TMRh20, 2016-04-11:
The example works, just some of the comments are wrong & make it confusing.

It only uses the watchdog timer, not interrupts to wake. To wake using interrupts, just remove stopListening() and see [sleepNode()](http://tmrh20.github.io/RF24Network/classRF24Network.html#a488bfa788f40b51bd5110b4594cf2d9a)

I think what happened, is I was experimenting with using ack-payloads to wake, and never completed it properly, so the example is all messed up.

aaddame, 2016-04-11:
Ok, that's how I got it working. Would be nice though if it works with
ack-payloads I guess

The example works, just some of the comments are wrong & make it confusing.

It only uses the watchdog timer, not interrupts to wake. To wake using
interrupts, just remove stopListening() and see sleepNode()
<http://tmrh20.github.io/RF24Network/classRF24Network.html#a488bfa788f40b51bd5110b4594cf2d9a>

I think what happened, is I was experimenting with using ack-payloads to
wake, and never completed it properly, so the example is all messed up.

acejoh, 2016-11-17:
I just ran into this issue as well, another vote to fix the documentation :)
So it would be possible to sleep the radio if you'd send an ack-payload first and then the real message to the sleeping node? Or it there more to it?

TMRh20, 2016-11-17:
The ack payload thing is just something I was experimenting with. 

Basically, if using ack payloads, you can do something like this:
1. Enable interrupts for the radio interrupt pin
2. radio.startFastWrite(); to send a payload
3. Immediately sleep the MCU
4. If the recipient responds with an ACK payload, the sleeping node will wake up via interrupt
5. The reason this works, is because the radio is left in TX mode, and internally the RF24 chip switches to RX mode and receives tha ACK payload, thus triggering an interrupt.

It many be possible that the same thing can be achieved as long as ESB/AutoAck is enabled but I have only tried it with ACK payloads.

This is not really useful in RF24Network, but may be useful in other configurations or specific implementations.

Plannign to look at the docs soon...

acejoh, 2016-11-24:
Oh right, I still don't get it but it's not so simple then :p 
Thanks for the reply and your continuing work on the library!

closed 2019-05-10 by TMRh20