nRF24/RF24Network #89: Arduino fragment dropped "out of order"

dan8551, 2016-04-12:
I have a raspberry pi (address 00) interfacing with an Arduino Mega 2560 (address 01). I am able to send a message from the pi to the arduino once without problems, however, subsequent transmissions result in fragments out of order. 

Any messages sent from the Arduino to the Pi are sent without issue.
Once the Arduino is reset the first message can be resent without problems and all subsequent messages are dropped (ACK is OK).

See debug outputs below:

First message received without problems
```
14627: MAC Received on 5 id 1 from 00 to 01 type 148
 14628: NET message 0409

14630: next_frame 1121 
14635: frame_queue 1121 
14639: NET Enqueue @0 
9
4
FA
FF
67
65
74
42
61
74
74
65
72
79
56
6F
6C
74
61
67
65
0
0
0
14658: MAC Received on 5 id 1 from 00 to 01 type 150

14667: NET message 0000

14671: next_frame 1121 
14675: frame_queue 1121 
14680: NET Enqueue @0 
fq 3: 36
9
4
FA
FF 
67
65
74
42
61
74 
74
65
72
79
56
6F 
6C
74
61
67
65
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
enq size 36
14708: NET message size 36
14713: NET r message 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 65 67 61 74 6c 6f 56 79 72 65 74 74 61 42 74 65 67 ff fa 04 09 

14736: next_frame 0 
```

Subsequent messages:
```
25751: MAC Received on 5 id 1 from 00 to 01 type 148

25751: NET message 2e54

25753: next_frame 0 
25756: frame_queue 1121 
25761: NET Enqueue @fb9f 
25765: MAC Received on 5 id 1 from 00 to 01 type 150

25775: NET message 0000

25779: next_frame 0 
25783: frame_queue 1121 
25787: NET Enqueue @fb9f 
Drop frag 0 Out of order 
```

If I comment out the code
```
if(  frag_queue.header.reserved == 0 || (header->type != NETWORK_LAST_FRAGMENT && header->reserved != frag_queue.header.reserved ) || frag_queue.header.id != header->id ){
			#if defined (SERIAL_DEBUG_FRAGMENTATION) || defined (SERIAL_DEBUG_MINIMAL)
			Serial.print(F("Drop frag ")); Serial.print(header->reserved);
			//Serial.print(F(" header id ")); Serial.print(header->id);
			Serial.println(F(" Out of order "));
			#endif
			return false;
		}
```

The messages are received without problems.

TMRh20, 2016-04-12:
Hey, can you post your code or an example?

dan8551, 2016-04-12:
```

#include "Arduino.h"
#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

RF24 radio(48, 53);                // nRF24L01(+) radio attached using Getting Started board
RF24Network network(radio);      // Network uses that radio
const uint16_t RPi = 00;    // Address of our node in Octal format ( 04,031, etc)
const uint16_t arduino = 01;   // Address of the other node in Octal format

struct payload_t 	// Structure of our payload
{
	unsigned long ms;
	char data[32];
};

void setup_nrf2401(void)
{
	SPI.begin();
	radio.begin();
	network.begin(/*channel*/ 90, /*node address*/ arduino);
	radio.setPALevel(RF24_PA_MIN);
	radio.printDetails();
}

void loop_nrf2401l(void)
{
	network.update();

	while ( network.available() )      // Is there anything ready for us?
	{
	  char data[32];
	  RF24NetworkHeader header;        // If so, grab it and print it out
	  payload_t payload;
	  network.read(header,&payload,sizeof(payload));
	}
}
```

TMRh20, 2016-04-12:
I'm not able to recreate any issues guessing on the tx code, but if I print out next_frame and frame_queue before queuing, they are always 0.

closed 2019-05-10 by TMRh20