nRF24/RF24Network #94: Intermittent issue with arduino uno

dekundu, 2016-05-22:
Hi,
I am trying to use RF24 mesh library to create a mesh and communicate amongst the devices. 
I used two examples, a)RF24Mesh_Example_Master_To_Nodes, and b)RF24Mesh_SerialConfig. Both the devices are connected to arduino uno. Now I see a strange problem sometimes with one side. I describe it below.

In the code I am opening serial port at 9600 baud, and then printed the "Setting up master at node 0" at the very beginning. It runs fine, but when I restarted/reconnected the arduino next day, the serial output did not come correct, it showed "Setting " an then some garbled up character. Please note, between these two runs I did not change the code. Now I reset the arduino quite a few times, but found different types of garbled output. Then I restarted the pc, reconnected the arduino in usb (did not rewire the radio though). Then I loaded the getting started program with radio.printDetails() turned on. It shows radio data successfully. Then I again reloaded the Mesh master program, and found to be working.

The other arduino does not seem to have similar problem. But both works fine while using the getting started program.
My question is whatever be the case the Serial output some come correct right.
Is the arduino having some problem or the library?
Thanks
Debojit

TMRh20, 2016-05-22:
What IDE version? If 1.6.8 please try updating it.

dekundu, 2016-05-23:
I am using arduino 1.6.9. Is it really related to arduino version? 
I am stuck with this intermittent behavior.
Please help.

TMRh20, 2016-05-27:
I don't think so if using 1.6.9, but I haven't had a chance to attempt to recreate. Sounds like some sort of memory issue, with the lib or the Arduino, but I'm not sure.

closed 2019-05-10 by TMRh20