nRF24/RF24Network #96: network.write() returns incorrect result

andreypopov, 2016-05-26:
Node 00 is master on RPi2
Node 01,011-051 send data to 00

Sometimes network.write() returns incorrect answer: false while 00 receives message  

Here is the screenshot of this bug (051 sends data to 00):
(as you see, I sent command "stop" 3 times, after stop fires "stat" command, the third stat command failed, but RPi received it)

[image]

nrf24 device settings are very common:
```
  radio.begin();
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_250KBPS);
  radio.setAutoAck(1);
  radio.enableAckPayload();
  network.begin(/*channel*/ 108, /*node address*/ this_node);
```
I tried to add delays between commands - nothing helps.

Please, help

TMRh20, 2016-05-27:
Without looking at this closely, I assume this is just a problem with ACKs, but maybe not an 'issue' per se...

Whenever a packet is routed, the final routing node sends back a 'network ack' which is software driven. When sending directly, radio auto-acks are used instead. 

There is always a possibility that a packet will be received and not acked, but it shouldn't happen very regularly, unless there is congestion. One thing to keep in mind, is that payload types 1-64 wil not be acknowledged via software, while user types 65-127 will be. If expecting a response, use types 1-64. All in all, if you want assured delivery, you will have to deal with duplicates.

If you think there is more of an issue here than I suggest, please advise.

andreypopov, 2016-05-27:
1-3 devices work almost perfect "failed" very rare, 4-6 - like 50%  of failed.
[image]

I didn't understand what do you suggest, because I am new one in electronics and my native language is not english)
I just want to find the reliable way to send small (<32 bytes) text message, even when there are 10 devices sending data at once.

Avamander, 2016-05-28:
Just asking, are you sure you have original nRF24s? The clones tend to have
intermittent issues like that.

andreypopov, 2016-06-04:
Chinese nrf24.. where to get the best nrf?

mdundek, 2016-06-28:
Hello all,

Same problem here, I have a rPI root node (00), one Arduino 01 node and one Arduino 011 Node. 01 is sending 6 messages at once every 30 seconds to 00, no problems there, all get acknowledged. Node 011 sends one message every 20 seconds to 00 and network.write returns false 4 times out of 5, but the messages DO arrive at node 00. Distance is not a pb, they are all on the same table. When I change address of 011 to 02, all works as expected.
I have used those modules before, this is the first time I encounter this behaviour, any advice would be welcome.

Michael

tguillaume02, 2017-09-14:
Hello all,
@andreypopov Have you found a solution for your problem? 
I have the same.
I use library nrf24 and nrf24Network
I have nodes 0, 01 and 011.
0 sends to 011 and sometimes 011 doesn't received. 
And sometimes 011 receive and send response to 0 but  0 didn't recevied response.

Thank you

andreypopov, 2017-09-14:
@tguillaume02 yes! I dont use this library anymore) MySensors lib works much more stable. And it is simplier.

tguillaume02, 2017-09-14:
ok thank you

Avamander, 2017-09-14:
If a packet isn't received and both CRC/ACK and re-transmissions fail (you can increase that count), it's up to you to deal with the error, that's the nature of radio communication, if you can't handle lost packets, yes, please do use some other library. RF24* is meant to provide the bare minimum you need to get started not some full-blown system that always performs flawlessly, is fast and lightweight (which is impossible). If you do receive a packet and ACKing it fails, too bad, that's again where you have to design your protocol properly to deal with those issues..

Disclaimer: If you find a flaw in RF24Network that's not related to your hardware (every possible hardware fault case is excluded), PLEASE actually report it so it can be fixed, so far I've seen **very few** that aren't actually related to poor hardware setup or clones.

tguillaume02, 2017-09-14:
ok, I don't know if the problem is due to module because if I test comunication beetween 0 and 011 ( change in 1 ) I have not problem.
Maybe my configuration : 
Master ( Node 0 )
```
#include <Arduino.h>
// ADD FOR NRF24L01+
#include <SPI.h>
#include <RF24.h>
#include <RF24Network.h>
#include <ReadSerial.h>  // include for read data send by USB for execute action on other node sended by SendDataRF24
const uint16_t this_node = 0;

// Structure of our payload
struct payload_t {                  
	int payload_pinId;
	int payload_value;
	int payload_etat;
	int payload_type;
};

#if defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny84__)
	//attiny84
	RF24 radio(8,7);
#else
	RF24 radio(9, 10);
#endif

// init for RF24NETWORK
RF24Network network(radio);
//RF24NetworkHeader header;

void SendDataRF24(int pin, int value, int etat, int other_node,  int type = -99)
{
	payload_t payload = { pin, value, etat, type };
	RF24NetworkHeader header( other_node,  'S');
	network.write(header, &payload, sizeof(payload));
}

void Check_RF24()
{
  while ( network.available() )
	{  
    payload_t payload;
    RF24NetworkHeader header;
    network.read(header,&payload,sizeof(payload));
    if (this_node != 0)
    {
        //analogWrite(IpinRF24, atoi("255"));
        digitalWrite(payload.payload_pinId,payload.payload_etat);
        RF24NetworkHeader header(/*to node*/ other_node, /*type*/ 'S');
        network.write(header,&payload,sizeof(payload));
    } 
    else if (payload.payload_value >= 0)
    {
        #if !defined(__AVR_ATtiny85__) && !defined(__AVR_ATtiny84__)
            Serial.print(String(payload.payload_type)+"_"+String( payload.payload_pinId)+"@"+payload.payload_value/100+":"+payload.payload_etat+"/");
            Serial.print("0");
            Serial.println(header.from_node,OCT);
        #endif
    }
  }
}

void setup()
{  
  radio.begin();
  sensors.begin();
  radio.setDataRate(RF24_250KBPS);  
  network.begin(this_node);  
  attachInterrupt(0, interruptHandler, CHANGE);
  difference = millis();
  difference1 = millis();
  Serial.begin(115200);
}

void loop()
{
  network.update();  
  Check_RF24();  
  ReadSerial();
}
```

Avamander, 2017-09-14:
I'd enable network debugging and see if ACK is sent and if it is where the ACK gets lost. I can't debug that for you (You can use Serial.print() in functions inside the library to see what's going on).

tguillaume02, 2017-09-14:
how enable network debugging and see if ACK is sent ?
On master or on the node 011 ?

Avamander, 2017-09-14:
The configuration file contains all the flags you could use. Define same flags on all nodes on the network.

tguillaume02, 2017-09-14:
ok thank you

closed 2019-05-10 by TMRh20