nRF24/RF24Network #97: I can't set the required topology

Hassan-Ali-Hassan, 2016-05-26:
Hi everybody ... First, I'd like to thank you TMRh20 for this great work. I apologize in advance because I'm still a newbie so my question may seem trivial, but I'm really having a hard time and I really appreciate any help.

 I've been trying to use the RF24mesh class in my swarm robotics project. I have a ground station that consists of a camera system that estimates the robots locations and a RF24 master node. Each robot in the system must know its location, as well as the locations of its neighbors (a neighbor of a node is other robot that has a pre-defined explicit link as defined by the network topology). My plan is to send each robot its location as estimated in the ground station and let the agents exchange their locations with each other.

The problem that is this plan requires all nodes communicate with the ground station and with each other almost simultaneously which leads the robot nodes to hang and stop receiving data. 

My question is: Is there any way to regulate the communication among agents and the ground station so as not to cause this jamming?

I attached the two files I use on master and slave nodes in case it helps. Thanks in advance.

[master.txt](https://github.com/TMRh20/RF24Network/files/283956/master.txt)
[slave.txt](https://github.com/TMRh20/RF24Network/files/283958/slave.txt)

TMRh20, 2016-05-27:
You might have better luck just using the RF24 library, and use a single broadcast address for every robot node, else you will be dealing with delays due to routing and network management.

RF24Mesh is designed as a 'sensor reporting' type network, and it operates most efficiently with all data flowing in the same direction. (master -> nodes  OR nodes -> master ) with more of a focus on self-maintenance and ease of use than for an advanced control system like this.

The biggest challenge is that RF24 radios are [half duplex](https://www.google.ca/search?q=half+duples&oq=half+duples&aqs=chrome..69i57j0l5.1560j0j4&sourceid=chrome&ie=UTF-8#q=half+duplex) so data collision is problematic.

ie:
1. Have all robots listen using the same address and turn off autoAck ( limits distance for all nodes )
2. Have the master node listen on a separate address and turn off autoAck
3. Jam as much information into a payload as you can, with instructions for each robot, since each one will receive the same data
4. Have robots report position to master, while master continually broadcasts information maybe?

This is just a suggestion, as there may be a few good ways to go about this.

Hassan-Ali-Hassan, 2016-05-28:
Thanks for your reply, I really appreciate it. It seems that I have two options now that will need some tweaking:
1) As you said, jamming all the data in one payload packet and send it to all agents to be parsed. 
2) Looping on all agents by the master node and sending to each node its location and the locations of its neighbors only.

The first idea has some advantages like the code simplicity of the master node, and that the data will be updated for all agent nodes simultaneously which is a good thing for the control algorithm (which doesn't like non uniform sampling rates). However it will increase the complexity in the agent nodes (now they have to parse the data and figure out who the neighbors are) which will cause the code to be customized for each agent (will be too annoying in testing phase). Also scaling up the system (by increasing the number of agents) will lead to bigger data packet which will take longer time to deliver to each agent.

The second idea also has some advantages: obviously the code will be simpler for nodes (each agent will only have to take the supplied data knowing that the extra data is for its neighbors eligible for it to communicate with) and also the data packets will be of almost the same size. However, scaling up the system will cause the number of packets to be sent to increase and this means that the for loop updating the locations of each agent will take longer time, so agents won't be able to simultaneously update their location, which will not be so good for the control algorithm. 

I will try the two approaches as soon as I can and update you with the results. 

Thanks again for your help.

closed 2019-05-10 by TMRh20