nRF24/RF24Network #98: Problem with multicast

ciberkids, 2016-08-02:
Hi,
I have an issue with multicast and the RF24Mesh that make use of
RF24Network's multicast features, after a very deep reverse engineering of the code I figured out why the multicast isn't working.
Unfortunately I haven't enough experience to understand if my workaround could work.

The problem reside on how the packet is composed during the calling fall of the various writes.

In the function _**write_to_pipe**_ there are these instructions
````
if (multicast) { radio.setAutoAck(0, 0); } else { radio.setAutoAck(0, 1); }

  radio.openWritingPipe(out_pipe);

  ok = radio.writeFast(frame_buffer, frame_size, multicast);
````
Here the multicast flags if brought to the underlying layer: RF24 lib
Going deeper, the first called function is:
`````
bool RF24::writeFast( const void* buf, uint8_t len, const bool multicast )
`````
This function sets some bits on the register of the module and it doesn't care about the flag and pass it to the **second _writefast** function_ that is:
`````
void RF24::startFastWrite( const void* buf, uint8_t len, const bool multicast, bool startTx){ 
````

In this function the flag changes a variable for the next function 
`````
	write_payload( buf, len, **multicast ? W_TX_PAYLOAD_NO_ACK : W_TX_PAYLOAD** ) ;
````
After one day of tests i figured out that the value **W_TX_PAYLOAD_NO_ACK** doesn't work.
In fact if i use the function:
`````
    network.multicast(header, &payload, sizeof(payload), 01); //Using the level 1 my parent node is 1

````
Into my arduino sketch and the right topology, nothing is received by the other arduino on the multicast pipe, nothing is received at all actually.

sender log for one send:

`````
RF24Network::multicast header: id 1 from 011 to 0100 type 129
RF24Network::multicast message size: 9 to level: 1 LevelToAddress: 1
RF24Network::write (0) 
RF24Network::write (0) header:id 1 from 011 to 0100 type 129
RF24Network::write (0) len:9 max_frame_payload_size: 24
RF24Network::_write (1) 
31350: NET Sending id 1 from 011 to 0100 type 129
Len of message: 9 writeDirect:(OCT) 1 writeDirect:(DEC) 1 writeDirect:(HEX) 1
RF24Network::_write (1) writeDirect != 070
RF24Network::_write (1) header type became USER_TX_MULTICAST
RF24Network::write (1) write(writeDirect= 1, sendType= 4 )
RF24Network::write (2)
RF24Network::write ACK requested for this frame
RF24Network::write to_node= 1 direct_to: 4
RF24Network::logicalToPhysicalAddress Multicast so pipe 0
RF24Network::logicalToPhysicalAddress to_node: 1 directTo: 0
31441: MAC Sending to 1 via 1 on pipe 0
RF24Network::pipe_address dec: 1
initial pipe: 0CCCCCCCCCC after cicle: 0CCCCCCCCCC else[1] -> 0x0CCCCCC3CCC
31467: NET Pipe 0 on node 1 has address 0x0CCCCCC3CCC sizeof(result): 8
RF24Network::write_to_pipe to node: 1 via pipe: 0x0CCCCCC3CCC multicast: YES
RF24RADIO STOP listening
write_register(00,0e)
write_register(02,3f)
write_register(01,3e)
write_register(11,20)
RF24::writeFast multicast flag:true
RF24::writeFast() start Writing
RF24::startFastWrite() start Writing
RF24::startFastWrite() Multicast flag:true startTX flag:true 
RF24::write_payload()
payload_size: 32
[Writing 17 bytes 15 blanks]
RF24::write_payload() writeType176
RF24::write_payload() status
14
RF24Network::write_to_pipe ok: YES
RF24Network::write_to_pipe !(networkFlags & FLAG_FAST_FRAG) 
write_register(01,3e)
RF24Network::write_to_pipe second chance ok: YES
Wrote! i hope
31597: send OK 
RF24Network startListening
RF24RADIO START listening
write_register(00,0f)
write_register(07,70)
````

receiver log for that send:
`````
................... nothing ..................
``````

Instead, if I use the raw pipes (exactly the same address of the multicast pipe) and the same payload and header used before directly with the radio library without the RF24Network on top I have all the communication working.

`````
  const uint64_t pipes = { 0xCCCCCC3CCCLL};

 memcpy(network.frame_buffer, &header, sizeof(RF24NetworkHeader));
      memcpy(network.frame_buffer + sizeof(RF24NetworkHeader), &payload, sizeof(payload));
      radio.openWritingPipe(pipes);

      bool ok = radio.writeFast(network.frame_buffer, sizeof(network.frame_buffer));
      if (!ok)
        Serial.println("e sti cazzi");
      if (!(network.networkFlags & FLAG_FAST_FRAG)) {
        ok = radio.txStandBy(network.txTimeout);
        radio.setAutoAck(0, 0);
      }
````` 

**This configuration turn on the flag **W_TX_PAYLOAD**. And in this case all is working.
Now my question is: if the ack is already turned off by the 
**setAct(0, 0)**(see above)
 is possible to use only the flag W_TX_PAYLOAD instead of W_TX_PAYLOAD_NO_ACK?**

log for one send:
`````
payload dimensions:9
header dimensions:8
framebuffer dimensions32
RF24RADIO STOP listening
write_register(00,0e)
write_register(02,3f)
write_register(11,20)
RF24::writeFast multicast flag:false
RF24::writeFast() start Writing
RF24::startFastWrite() start Writing
RF24::startFastWrite() Multicast flag:false startTX flag:true 
RF24::write_payload()
payload_size: 32
[Writing 32 bytes 0 blanks]
RF24::write_payload() writeType160
RF24::write_payload() status
46
write_register(01,3e)
````

receiver log for that send:
````
trying to read from radio
During RF24NETWORK update read bytes:32
3516398: MAC Received on 0 header id 1 from 011 to 01 type 4
RF24Network::update test: header->to_node->1==1-->node_address
RF24Network::update it is for us!
3516426: NET Enqueue @0 3516430: NET Enqueue0
IT IS FOR MEEE!!!
3516437: NET message size 24
3516441: NET r message 01 00 00 00 12 00 01 d1 9e 

header:id 1 from 011 to 01 type 4
Received packet #18 at 119198 with multicast flag on payload: YES
````

sorry for the long writing but I neede to be as clear as possible.
Best regards Matteo

Avamander, 2016-08-05:
Hi, can you please try and obtain some 100% genuine modules and then test again?

TMRh20, 2016-08-07:
Thanks for all the details & info on this issue.

> This configuration turn on the flag W_TX_PAYLOAD. And in this case all is working.
** Now my question is: if the ack is already turned off by the 
**setAutoAck(0, 0)(see above)
is possible to use only the flag W_TX_PAYLOAD instead of W_TX_PAYLOAD_NO_ACK?**
  
@ciberkids Yes. 

This all became very confusing when we identified that clones seem to have problems with the optional no-ack flag, so that is why ACKs are now disabled on a specific pipe instead of using dynamic-acks.
  
I haven't fully cleaned up/reviewed/tested the code from that change, but you could try changing the line ``` ok = radio.writeFast(frame_buffer, frame_size,multicast); ``` to ``` ok = radio.writeFast(frame_buffer, frame_size,0); ``` which would just leave the multicast flag off.  
  
Please let me know if this works or put in a pull request if so, as I assume now that it may fix some compatibility problems with more clones.

ciberkids, 2016-08-07:
Hi to all, 
@Avamander as soon as I'll receive the other radios(that should have the original chip)  I'm going to try all so I'll tell you if it is an un-implemented features,  but I think you are right. 

@TMrh20
Thank you for you answer I did a lot of changes on the libraries to have both implementation working ( with flag and without)  at the moment my radio is working with with the flag off (my implementation not your suggestion because I moved forward before your answer :) :) ) .  Moreover I implemented a better system for getting a multicast per level message,  using a more specific value for each multicast level (not only 0100 for the "write direct" option but a address like 0400,0300, 0200,0100 that represents the level where you want to multicast to). 
Last but not least,  during my re-implementation I saw a lot of part where a refactoring is possible but sometime it breaks the original api (I kept all old but I add some new,  it should be backwards compatible) . 
The refactor consists in a better reading,  adding some more documentation,  because sometimes it isn't very clear how to use it,  and cleaning...  when I'll finish I'll push all the code to my repo and then I'll ask for a pull request....  if you will think that the changes are too much you can refuse:) 

ps.  now I am on holidays so I can't touch the code...  but in few days I'll back and I'm going to finish the remaining work.

TMRh20, 2016-08-08:
Sounds good!

Since testing is OK and you confirmed my suspicion, I committed the change to remove the W_TX_PAYLOAD until your changes are ready. It would be great to have somebody help clean this lib up a bit, so I'm looking forward to your changes.

closed 2019-05-10 by TMRh20