nRF24/RF24Network #100: undefined reference to `RF24Network::setup_watchdog(unsigned char)'

romandubrov, 2016-11-06:
This compile error happens when #define ENABLE_SLEEP_MODE line is commented out in RF24Network_config.h

The reason is pretty simple: there are two functions (setup_watchdog and sleepNode) that wrapped into #if defined ENABLE_SLEEP_MODE in .cpp file but not in .h file. I've wrapped their definitions also and it has compiled successfully.

#if defined ENABLE_SLEEP_MODE
 bool sleepNode( unsigned int cycles, int interruptPin );
#endif

#if defined ENABLE_SLEEP_MODE
 void setup_watchdog(uint8_t prescalar); 
#endif

TMRh20, 2016-11-07:
Something seems odd with this. 

When creating a library, for example, you can define all of your functions in the *.h file without creating them in the *.cpp files, and it will compile just fine.

As long as ENABLE_SLEEP_MODE is actually commented via `` // `` it should be fine??

romandubrov, 2016-11-07:
@TMRh20 well it should compile ok but linker will throw an error. If you google a bit, you'll find lot of people who got this issue and had to uncomment that define.

TMRh20, 2016-11-07:
The only thing I really find on google involving a solution links back here: https://github.com/TMRh20/RF24Network/issues/84

A couple questions:
What platform are you getting this error with? Arduino, Linux/RPi ?
What IDE? Arduino?
Have you previously installed another version of RF24Network?
Can you try removing any and all RF24Network files and reinstall the library?

It works fine for me on Arduino, Linux, ESP/NodeMCU, etc

romandubrov, 2016-11-07:
I have latest arduino IDE with stock latest RF24* libs. yes, tried to remove/add - no effect

TMRh20, 2016-11-07:
Ok, try this:

1. Open your Arduino/libraries/RF24Network/RF24Network_config.h
2. If you do see this exactly ``` //#define ENABLE_SLEEP_MODE ``` with it
commented out, then do the following
3. Using Arduino library manager, remove RF24Network
4. Go back to the Arduino/libraries/ folder and delete RF24Network folder
5. Reinstall using Arduino Library manager

There is no reason it works for me but not you, other than something else
being messed up.

romandubrov, 2016-11-07:
bingo! 
it seems arduino's built-in library manager doesn't replace all files (I wonder why)
deleting a whole libraries/RF24Network folder solved the problem
I should be more rigorous
thanks

closed 2016-11-08 by TMRh20

pyashwanth32, 2020-12-05:
Thank you so much!!
I was facing the same problem. 
Go to Sketch -> Include Libraries -> Manage Libraries. 
In the Library manager, search 'RF24' and from the drop down list, Install all the libraries that say RF24. 
Compile the Code again and it worked for me!
Thanks