nRF24/RF24Network #101: 2 way communication?

kartakur, 2016-11-17:
Is there a solution for full-duplex communication? I mean without doing radio.listen, radio.stop over & over again, I want both transmitter & receiver modules remain open..Is it possible or I have to use 2rf module for that. 
Btw I'm trying to do transmitting image(2fps is enough) & receive control commands at the same time but radio.stop-radio.listen makes too much delay..

(Sorry for english)

Avamander, 2016-11-17:
No, full duplex is not possible, but you can use two transceivers.

TMRh20, 2016-11-17:
The closes thing to full duplex with a pair of radios is letting the radios manage the TX/RX switching internally. (bidirectional  half-duplex communication)

You can do this using AckPayloads. See the [example](http://tmrh20.github.io/RF24/GettingStarted_CallResponse_8ino-example.html)

kartakur, 2017-01-01:
Thanks for reply. I have one more question :) Can I use dual head mode on RPi. 

I have 2 rf module (total 4, 2 per RPi) and I can use RF24 library to communicate between 2 RPi like full-duplex(using one of modules only reciever and other one transmitter) but it felt so slow. So I want to use RF24Network because at least I can send more byte per package. Can you plase tell me how I can use dual head if it's possible and if it's not what can I do to achieve at least 250 kb/s (kilobyte per second) data transfer speed.

closed 2019-05-10 by TMRh20