nRF24/RF24Network #104: Extend header

ppisljar, 2017-01-12:
Hi, 

I am looking for a way to add additional field to header.

i tried by simply defining another uint16_t msg_id; to the .h file  and it seems to break communication completely. I was looking thru the code and I see that in many locations header size is hardcoded ? its not checking the size of RF24NetworkHeader in all locations ... so i tried to update that ... now it fails even worse :)

Seems I don't fully understand how the packet is constructed.

to my understanding the package (for example in enqueue function) is constructed with : 
- 8 bytes header - 2 bytes msg length - payload (max 22 bytes)

so with my change i would like:
- 10 bytes header - 2 bytes msg length - payload (max 20 bytes)

i updated every 8 in the code with sizeof(RF24NetworkHeader) and every 10 with sizeof(RF24NetworkHeader) + 2; for example this code:

```
memcpy(next_frame,&frame_buffer,8);
memcpy(next_frame+8,&message_size,2);
memcpy(next_frame+10,frame_buffer+8,message_size);
```

was updated to :
```
memcpy(next_frame,&frame_buffer,  sizeof(RF24NetworkHeader));
memcpy(next_frame+ sizeof(RF24NetworkHeader),&message_size,2);
memcpy(next_frame+ sizeof(RF24NetworkHeader)+2,frame_buffer+ sizeof(RF24NetworkHeader),message_size);
```

is there some place in code that my assumption is wrong ? and am i understanding how message is constructed correctly ?

ppisljar, 2017-01-12:
I would love to help update the original code to correctly use `sizeof(RF24NetworkHeader)` in every place so extending the header would be just a single line of code.

Avamander, 2017-01-12:
It was probably hardcoded to make the size more apparent and easier to
understand to anyone modifying the code. What's your goal of editing the
header anyway?

ppisljar, 2017-01-12:
i want to add message id to the header (so that i can know which reply is to which message)

Avamander, 2017-01-12:
Why not include that in the payload?

ppisljar, 2017-01-12:
Yes, you are right. Not really sure why, but it looked better to me to get that data in the header (as well as the size of payload maybe). However more that im thinking about it it seems this should be in the payload (to avoid resending with every fragment)

Avamander, 2017-01-12:
In the end it'll just be easier for you because you can then just update the library and the code you wrote keeps working. I seriously doubt there will be huge API-breaking changes and when there will be a next major version you can simply choose not to update.

closed 2019-05-10 by TMRh20