nRF24/RF24Network #105: losing the last few packets when having many fragments because tx is too fast

mamama1, 2017-01-15:
Hi

when I send about 544 bytes of data including header (= 17 frames @ 32bytes), the last 2 fragments get lost. I've enabled debugging and I can see that on the receiver side nothing is coming in.
the last debug message on the receiver side is:

`187723 MAC: System payload rcvd 149
187726: NET Enqueue @0
187729: MAC Received on 5 id: 6 from: 05 to: 00 type: 149
187734: NET message 3635
187736: dnX frame_buffer raw: [5 0 0 0 6 0 149 2 68 78 88 78 79 68 69 48 49 53 54 55 56 57 48 49 50 51 52 53 54 55 56 57 ]`

so we are missing frame 1 and 0. besides, the last frame should have type 150.
on the sender side, we can see this:

`
....
5295: MAC Sending to 00 via 00 on pipe 5
5299: NET Pipe: 5 on node: 00 has address: cccccccce3
5303: FRG message transmission with fragmentID '3' sucessfull.
5309: NET Sending id: 4 from: 05 to: 00 type: 149
5313: NET message
5314: MAC Sending to 00 via 00 on pipe 5
5318: NET Pipe: 5 on node: 00 has address: cccccccce3
5323: FRG message transmission with fragmentID '2' sucessfull.
5328: NET Sending id: 4 from: 05 to: 00 type: 149
5332: NET message
5335: MAC Sending to 00 via 00 on pipe 5
5338: NET Pipe: 5 on node: 00 has address: cccccccce3
5343: MAC Send fail to 00 via 00 on pipe 5
5348: FRG message transmission with fragmentID '2' sucessfull.
5352: NET Sending id: 4 from: 05 to: 00 type: 149
5356: NET message
5357: MAC Sending to 00 via 00 on pipe 5
5361: NET Pipe: 5 on node: 00 has address: cccccccce3
5366: FRG message transmission with fragmentID '1' sucessfull.
5371: NET Sending id: 4 from: 05 to: 00 type: 150
5376: NET message
5378: MAC Sending to 00 via 00 on pipe 5
5381: NET Pipe: 5 on node: 00 has address: cccccccce3
5386: MAC Send fail to 00 via 00 on pipe 5
5391: FRG message transmission with fragmentID '1' sucessfull.
5395: NET Sending id: 4 from: 05 to: 00 type: 150
5399: NET message
5401: MAC Sending to 00 via 00 on pipe 5
5404: NET Pipe: 5 on node: 00 has address: cccccccce3
5409: FRG message transmission with fragmentID '0' sucessfull.
5474: NET Sending id: 5 from: 05 to: 00 type: 130
5475: MAC Sending to 00 via 00 on pipe 5
5476: NET Pipe: 5 on node: 00 has address: cccccccce3
Send P fail, Test OK`

so this means that the sender is sending the data but the fragment IDs 2 and 1 already get retransmitted but eventually the sender thinks that they are being received. eventually mesh.write fails but mesh.checkconnection returns true (cause everything else still works) so we get the "Send P fail, Test OK" message.

I added several debugging messages and at some point suddenly the complete payload got delivered. I realized that if I add a delay of about 50ms to the RF24Network::_write(RF24NetworkHeader &header, const void *message, uint16_t len, uint16_t writeDirect) function or even to the RF24Network::write_to_pipe function, the payload gets delivered.

So it seems that somehow, when using huge payloads and thus many fragments, the input buffer of the receiving RF24 module gets filled up (or something like that). I don't know what the ultimate solution should be and I also didn't fully understand why the sender thinks that the last two fragments got delivered. are they hardware acknowledged and the receiver NRF acknowledged them without actually delivering them to the software? no idea.

I'm not planning to send payloads like this, I just was testing some changes and wanted to check fragmentation and delivery of huge payloads. 288 bytes (9 frames/fragments that is) are no problem without an added delay.

Platform is Arduino Mega for the Master node and Arduino Uno for the client node.
The issue arised in combination with the Mesh Library but as this only occurs when having a huge number of fragments, I guess the problem isn't located in the mesh library.

I've already bought Sparkfun NRF24L01+ modules which cost about $20 each because I already lost much time using counterfeits and searching for errors which turned out to be hardware related.
I suspect (and I HOPE) that those Sparkfun NRFs are genuine.

Any ideas? As I was studying and changing the RF24 libs a lot in the past weeks I think I can help getting a solution. I just need some directions please. The low level RF24 lib is something I didn't really look at yet because I'm quite new to electronics/AVR/embedded development and reading/writing registers and reading as well as understanding code which does that isn't just as my thing yet as reading/writing standard C (which I'm doing only for about 9 months as well).

Thanks!

Avamander, 2017-01-15:
Your assumption seems right, the receiving side's buffer gets filled.
Though @TMRh20 is more familiar with the intricate details of the modules
and might be able to tell you exactly happens.

mamama1, 2017-01-15:
do you think its the NRFs buffer or the arduinos software buffer?

Avamander, 2017-01-15:
You can make the Arduino print out the free memory and even try to
temporarily malloc(), then free() to make sure it actually can allocate
enough memory to buffer another payload. To know if it's the NRF, see the
datasheet.

mamama1, 2017-01-15:
thanks for your reply!
all buffers i was looking at so far were statically assigned, so no malloc. maybe there's something in the RF24 lib. in RF24Network I didn't see anything like that.

mamama1, 2017-01-15:
after some more investigation I can say that no malloc/calloc is done by RF24 or RF24Network. Also the address list is of a fixed size in my modified RF24Mesh library so in the whole project there is no dynamic memory allocation done.

I had a pretty bunch of debug messages printing out on the receivers side. when I turned some of them off and removed the 50ms delay, transmission of huge payloads with many fragments still worked.

so for me it looks like the RX buffer of the RF24 modules has congestion and as it is a FIFO buffer, the data which isn't collected by the MCU in time gets pushed out and is lost.

this would imply that the RF24 is sending its hardware auto-ack as soon as it received the packet, regardless of the MCU collecting it which is quite bad for me right now.

The debug messages won't be there when the project is finalized but the MCU has a bunch of other jobs to do and I'm afraid that at some point data will get lost again, although ACKed by the NRF modules (on which the RF24 libs completely rely on in some points).

This means, that I'd have to implement some kind of software ACK on top to ensure packets are being transmitted.
But then again, I'm not sure why eventually mesh.write fails. Somehow, the software gets to know that something happened. Maybe some retries would already repair it?

@TMRh20 please help me with this one...

thanks!

closed 2019-05-10 by TMRh20

mamama1, 2019-05-10:
I have moved on anyway but I'm curious - any thoughts about this?

TMRh20, 2019-05-10:
I would have to see some code and experiment a bit. I've not encountered this myself, but the MCU should have no trouble emptying the FIFO buffers in time, as long as network.update() is being called regularly. I've done a lot of my testing using RF24Ethernet/RF24Gateway (TCP/IP uses large payloads) and have not encountered this problem.

mamama1, 2019-05-10:
Ok, thanks for coming back to me!