nRF24/RF24Network #107: RF24Network and NodeMCU ESP8266

ciapamusche, 2017-01-24:
Im tryng to run my sketch in which include rf24network library non my node mcu esp8266
Usually i run it in an ATmega without problem
i get this error message.

Arduino:1.6.13 (Mac OS X), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

```
/Users/Damiano/Documents/Arduino/libraries/RF24Network-master/RF24Network.cpp:30:24: fatal error: avr/sleep.h: No such file or directory
  #include <avr/sleep.h>
                        ^
compilation terminated.
exit status 1
```
i know the avr/sleep.h is not a ESP library but i don't have this library in my sketch.
with 
include RF24.h only it compile.

any advice ?thank you very much for this fantastic job... it works perfectly in my security system.

Avamander, 2017-01-25:
Can you enable the compilation debug and post the output of the full compilation?

ciapamusche, 2017-01-25:
Sure... here we are...

> 
> `Arduino:1.6.13 (Mac OS X), Scheda:"ESPino (ESP-12 Module), 80 MHz, DIO, 115200, 4M (1M SPIFFS), ck"
> 
> /Users/Damiano/Downloads/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/Damiano/Downloads/Arduino.app/Contents/Java/hardware -hardware /Users/Damiano/Library/Arduino15/packages -hardware /Users/Damiano/Documents/Arduino/hardware -tools /Users/Damiano/Downloads/Arduino.app/Contents/Java/tools-builder -tools /Users/Damiano/Downloads/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Damiano/Library/Arduino15/packages -built-in-libraries /Users/Damiano/Downloads/Arduino.app/Contents/Java/libraries -libraries /Users/Damiano/Documents/Arduino/libraries -fqbn=esp8266:esp8266:espino:CpuFrequency=80,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=ck -ide-version=10613 -build-path /var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.esptool.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -verbose /var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_modified_sketch_754457/RX_1.9.ino
> /Users/Damiano/Downloads/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/Damiano/Downloads/Arduino.app/Contents/Java/hardware -hardware /Users/Damiano/Library/Arduino15/packages -hardware /Users/Damiano/Documents/Arduino/hardware -tools /Users/Damiano/Downloads/Arduino.app/Contents/Java/tools-builder -tools /Users/Damiano/Downloads/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/Damiano/Library/Arduino15/packages -built-in-libraries /Users/Damiano/Downloads/Arduino.app/Contents/Java/libraries -libraries /Users/Damiano/Documents/Arduino/libraries -fqbn=esp8266:esp8266:espino:CpuFrequency=80,FlashMode=dio,UploadSpeed=115200,FlashSize=4M1M,ResetMethod=ck -ide-version=10613 -build-path /var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/mkspiffs/0.1.2 -prefs=runtime.tools.esptool.path=/Users/Damiano/Library/Arduino15/packages/esp8266/tools/esptool/0.4.9 -verbose /var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_modified_sketch_754457/RX_1.9.ino
> Using board 'espino' from platform in folder: /Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
> Using core 'esp8266' from platform in folder: /Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0
> Detecting libraries used...
> "/Users/Damiano/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10613 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/espino" "/var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666/sketch/RX_1.9.ino.cpp" -o "/dev/null"
> "/Users/Damiano/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10613 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266 "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/espino" "-I/Users/Damiano/Documents/Arduino/libraries/RF24Network-master" "/var/folders/s_/2yq07rw150v5yfj7z23hv0r80000gn/T/arduino_build_160666/sketch/RX_1.9.ino.cpp" -o "/dev/null"
> "/Users/Damiano/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/Damiano/Library/Arduino15/packages/esp8 [... cut ...]
> compilation terminated.
> Uso la libreria RF24Network-master alla versione 1.0.7 nella cartella: /Users/Damiano/Documents/Arduino/libraries/RF24Network-master 
> Uso la libreria RF24-master-2 alla versione 1.2.0 nella cartella: /Users/Damiano/Documents/Arduino/libraries/RF24-master-2 
> Uso la libreria SPI alla versione 1.0 nella cartella: /Users/Damiano/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI 
> exit status 1
> Errore durante la compilazione per la scheda ESPino (ESP-12 Module).

Avamander, 2017-01-25:
Update your library. Use the "Download as .zip" option, remove all other versions of the library.

ciapamusche, 2017-01-25:
Thank you  avamander for the reply.
Now it compile and load... my silly fault.... sorry

Now have another problem with the sketch. that i had not before. 

 i try to explain my situation in few words

My project is a sensor mash (atmega328 in stand alone 8mhz configuration) which send via RFnetwork a bool PIR state,a bool door contact state, a float battery state and a bool "battery low" flag...

all these message are in the message struct.

Usually the receiver was another ATmega328 in standalone configuration.

for debug i use serial.println on serial monitor for TX and RX which return me the part of payload of my interesting.

I noticed Which the value of battery (usually around 3.30...3.40 v in my case) is sent correctly by the transmitter but the receiver give me ever a completely wrong value... fixed to 0.50. and a "battery low flag" which normally can be 0 or 1 if the battery goes under 2,9 v) at 96 

with 2 Arduinos the sketch work perfectly...

can you help me?

i don't know if this is the right place to ask this question. If not you can close the question because for me the problem i solved updating RFNetwork library

Avamander, 2017-01-25:
Try multiplying the battery state by 100 (convert it to integer not float) and store it in a unsigned 8 bit variable (uint8_t), on all devices you have to update the struct with that change.

ciapamusche, 2017-01-25:
now the battery is seen 75 when its 315 and 0 when 0. The battery low flag seems work.

I cannot understand the problem...

i can to lets you see my sketches...

RX NODE ON ESP

```
`#include <RF24Network.h>
#include <RF24.h>
#include <SPI.h>

const uint16_t pi_node = 0; // Base
const uint16_t sensor_node1 = 01; // Primo ramo (parla solamente con 0 e con i suoi figli)
const uint16_t sensor_node2 = 021; // Secondo ramo (parla solamente con 01 e con i suoi figli)
const uint16_t sensor_node3 = 011;  // Secondo ramo (parla solamente con 01 e con i suoi figli)
const uint16_t sensor_node4 = 0121; // Terzo ramo (parla solamente con 021 e con i suoi figli)
const uint16_t sensor_node5 = 031; // Secondo ramo (parla solamente con 01)

// CE Pin, CSN Pin, SPI Speed
RF24 radio(2, 15); //impostare CE e CSN  (in questo caso pin 9 e 10);
RF24Network network(radio); 
// Time between checking for packets (in ms)
const unsigned long interval = 1000;

// Structure of our messages
struct message_sensor { 
 bool motion;
 bool dooropen;
 uin8_t Vbatt;
 bool s;
};

message_sensor message;

void setup ()

{

// Initialize all radio related modules

  Serial.begin(115200);
  delay(10);
  radio.begin();

  delay(5);

  network.begin(90, pi_node);

  radio.setPALevel(RF24_PA_MIN); 
  Serial.println("Receiver Ready"); 
  // Print some radio details (for debug purposes)
  radio.printDetails();
  network.update(); 
}
void loop()

  {
 bool goodSignal = radio.testRPD();

   if(radio.available()){

      Serial.println(goodSignal ? "Signal good > 64dBm" : "Segnale weak< 64dBm" );

       radio.read(0,0);

  }

network.update();
[... 300 lines cut ...]

{
 int sensorValue = analogRead(A0); 
                                
  Vpin_perc = map (sensorValue, 0, 1023, 0, 420); 
  Vbatt = Vpin_perc / 100 ;

  s=(Vpin_perc < 290); 
  Serial.println(s);
 Serial.println(Vpin_perc);
 // Serial.println(Vbatt);
 /* if (network.write(header, &messageo, sizeof(messageo))) {
      Serial.print("Battery Message sent\n"); 
    } else {
      Serial.print("Could not send Battery message\n"); 
    }
   
}
```

i can miss some ; in copy and paste....

the variable i send for battery is "Vpin_perc" and the bool is "s"

thank you for your help

Avamander, 2017-01-26:
You can't store value more than 255 (0-255) in uint8_t, change the map() in testbatt() to map it to maximum 255.

ciapamusche, 2017-01-26:
very thanks i tried to convert the float and sand but didn't work.. however with your advice transmitted an uint16_t to receiver with the value  and once received i divided by 100 in the receiver..

Thank you for your help. Question closed....

closed 2017-01-26 by ciapamusche