nRF24/RF24Network #112: network.update() freeze on Due

Antownee, 2017-04-28:
I have noticed that in my sketch, the network.update() function on the Arduino Due is not called. 
However, running the exact same sketch on my pro-mini works just as expected.

Anyone with some insight on this?
Thanks.

massiveradiation, 2017-07-14:
This is happening to me on an Uno clone, with the non Atmega chip. Just hangs at network.update().

Avamander, 2017-07-14:
Check your wiring, check your modules, check your pin definitions. See https://github.com/nRF24/RF24Mesh/issues/8#issuecomment-170259189 for more information.

massiveradiation, 2017-07-14:
Ah yes, I figured it out: the Rf24+ was attached to a breakout board, the common one with a built in 3.3V regulator. I was powering this board with a 3.3V voltage regulator attached to a wall wart. Not good. Just wasn't getting enough power. Once I provided her some juice, she came right back up. But I do have to say that I'm finding the RF24+ to be a very finicky piece of hardware when it comes to code. So far the RF24Networking library (Arduino) is the most solid way to connect multiple "nodes." I'll keep experimenting because I really want these to be more stable and predictable than they have so far proven. Thanks for chiming in Avamander, should I look into the mesh method/library? My needs are so simple, passing a single digit when a sensor is triggered.

Avamander, 2017-07-14:
If you have original (and quality) modules they rarely have these issues actually, but it's hard to find good suppliers from eBay. RF24Mesh is definitely worth looking into, it simplifies quite a few things (nodeIDs etc.) and is overall rather nice piece of software. But keep in mind RF24Mesh hangs if your modules lack proper auto-ACK and dynamic payload support so you might not even be able to use it.

massiveradiation, 2017-07-14:
Do you have a good source for either original or quality RF24+'s?

Avamander, 2017-07-14:
@massiveradiation unfortunately, no.

closed 2019-05-10 by TMRh20