nRF24/RF24Network #113: RF24 Network RPI 3

ggrilli75, 2017-05-16:
Hello
I'm trying to run RF24 Network in raspberri Pi 3, in python,

Follow the installation instructions for the RF24 and RF24 Network libraries

But when executing the example codes of the pyRF24Network folder, I deviated the following result:

   File "helloworld_tx.py", line 29, in <module>
     Radio = RF24 (RPI_V2_GPIO_P1_15, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ)
NameError: name 'RPI_V2_GPIO_P1_15' is not defined

Any idea to solve?
Thank you

TasPats, 2017-05-18:
need to add line:

`#include <RF24/utility/RPi/bcm2835.h>
`

flightjumpr, 2017-06-21:
Having the same issue on a pi zero :

Traceback (most recent call last):
  File "cpass_rpi_hub.py", line 215, in <module>
    radio = RF24(RPI_BPLUS_GPIO_J8_22, RPI_BPLUS_GPIO_J8_24, BCM2835_SPI_SPEED_8MHZ)
NameError: name 'RPI_BPLUS_GPIO_J8_22' is not defined

Avamander, 2017-07-14:
Clearly RPI_BPLUS_GPIO_J8_22 is not defined on a Pi Zero, replace the content with relevant and your Pi specific pins.

Avamander, 2017-07-14:
@ggrilli75 Did TasPats' suggestion help you?

vamsikrishnaA9, 2019-05-07:
@ggrilli75 i am also trying to run RF24 Network on RPI 3B+. When in try to run the example from pyRF24Network folder it is showing no module named RF24. 
Can you please share the installation details to run python script on rpi

ggrilli75, 2019-05-07:
Yes, I will send it to you.
to be able to help you as much as I can

vamsikrishnaA9, 2019-05-07:
Thank you very much for your reply. I am doing project with nrf modules
using rpi. But i am unable to run helloworld-tx.py file.
I am new to rpi linux environment please help

ggrilli75, 2019-05-07:
Whats error you  get up?

vamsikrishnaA9, 2019-05-07:
No module named RF24 
Not found

vamsikrishnaA9, 2019-05-07:
Actually i had seen to eliminate the above error we need to install python wrapper class https://tmrh20.github.io/RF24/Python.html see the link. After done steps same error is coming. I think i didn’t installed properly. If that is the only mistake correct me, if not please teach me the way to run python scripts

ggrilli75, 2019-05-07:
I made the following instructions for you...

1) Clone the RF24Network repository:
git clone https://github.com/TMRh20/RF24Network.git

2) Switch to the RF24Network folder
cd RF24Network

3) Clone there in the repository RF24
git clone https://github.com/TMRh20/RF24.git

4) Switch to the RF24 folder and compile the module
cd RF24
sudo make install
5) Go back to the folder to the RF24Network folder and compile the module
cd ..
sudo make install
**********************************************
********************************************** **********
Now comes the next stage of generating the python packages for RF24Network
and RF24

1) Run the python file setup.py in the folder RF24Network / RF24 / pyRF24
sudo python setup.py install
2) Run the python file setup.py in the folder RF24Network / RPi /
pyRF24Network
sudo python setup.py install

With these steps you all set to start working on python .... I hope you
found the information served.

vamsikrishnaA9, 2019-05-09:
hi, i have gone through above instructions you have made
Errors are coming at sudo python setup.py install command.
It is showing  boost/python.hpp: No such file or directory.
here is the screen shot which can describe more.
please lookinto this and help me
thank you
[image: 2019-05-09-083142_1600x900_scrot.png]
vamsi krishna

ggrilli75, 2019-05-09:
Do you know if boost packet is installed?

vamsikrishnaA9, 2019-05-14:
yes, i have installed and successfully run all python setup.py files. but i am unable to run the exact pyhton script files in python idle.( no module named RF24)
[image]
[image]
[image]

by sudo ./helloworld_tx.py i can able to run the program it is executed but pop with sum error.
when i run the same program through idle it won't.
i want that python scripts to be run idle so that i can write other code and use this network examples as a part in my project.
i am much thankful to your patience and listening to problem.

ggrilli75, 2019-05-14:
I see that you do not have the same library installed in my raspberry ...
If you want I offer to reinstall it remotely.

vamsikrishnaA9, 2019-05-15:
Thanks
my email id - vamsikrishnavuyyuru007@gmail.com

I will do my steps on pi please send me a mail.
-- 
With regards

Vamsi Krishna

closed 2020-05-27 by TMRh20

Younes-SadatNejad, 2020-10-21:
Hello, 
I am facing the same problem as the initial issue by ggrilli75. I get the error:
NameError: name 'RPI_V2_GPIO_P1_15' is not defined

I am using rpi 4 B model. 
Any help?

ggrilli75, 2020-10-22:
Hello, I would have to see if there is any variant for raspberry 4,
since I have implemented it with rpi 2 and rpi3.
Any news I notify you
Cheers

Younes-SadatNejad, 2020-10-22:
> Hello, I would have to see if there is any variant for raspberry 4, since I have implemented it with rpi 2 and rpi3. Any news I notify you Cheers El mié., 21 de oct. de 2020 17:55, Younes Sadat-Nejad < notifications@github.com> escribió:
> […](#)

Hello, 
Thank you for getting back to me. 
Is there any way around it?
I tried the following:

import spidev
spi = spidev.SpiDev()
spi.open(0, 0)
spi.max_speed_hz = 8000

radio = RF24(22,0)

The code runs without, but I do not receive anything .... 
I am sure the problem is not from the sender because Arduino to Arduino works fine. 

Please let me know?
Thank you
Best

2bndy5, 2020-11-02:
@Younes-SadatNejad If the adding that code worked for you, then the only line that would seem to do anything is the
```python
import spidev
```
Where as the rest of the code you posted would be problematic. I'm talking about this:
```python
spi = spidev.SpiDev()
spi.open(0, 0)
spi.max_speed_hz = 8000
```
because you're  essentially creating 2 object to manipulate *an* SPI bus: 1 with `spi = spidev.SpiDev()` and another with `radio = RF24(22, 0)`. It wouldn't hurt to find/read the documentation on the python wrapper for SPIDEV because I think `spi.open(0, 0)` is more applicable to RPi1 where `spi.open(1, 0)` is typical of RPi model B+ or later (going off of memory here so I could be wrong). IIRC, the second argument to `spi.open()` is the RPi's CE pin offering (can be 0 or 1).

I'm actually more curious if your code has this:
```python
import RPi.GPIO
```
as that is the library used for manipulating pins in the python wrapper. HINT: you may need to install the RPi.GPIO python module (via `pip3 install RPi.GPIO`). Also, if you're program isn't using python3, you should consider migrating to it as python2 is officially dead.

Beware of copying code from tutorials, especially if using GPIO pins and the tutorial is coded for antiquated device(s)/library(s)