nRF24/RF24Network #114: RF24Network Compatable with Arduino Zero?

ProblemSolved99, 2017-07-25:
Hi,

I am am working with a neat board (https://www.protocentral.com/analog-adc-boards/1143-protocentral-openpressure-24-bit-pressure-sensor-daq.html) that has a  Atmel ATSAMD21 micro-controller loaded with Arduino M0/Zero bootloader. 

I am trying to upload to the board with the Arduino Zero (Native USB Port) selected and Programmer Atmel-ICE selected as well. When uploading it throws the following warning:

"WARNING: library RF24Network claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [samd] architecture(s)."

Is this library compatible with samd?

Thank you.

Avamander, 2017-10-02:
Does it work?

ProblemSolved99, 2017-10-02:
Avamander,

Thank you for your response. It turns out that it was more of a warning message, but the board would still work fine. I ended up moving away from using that board so it became a non-issue for me.

closed 2017-10-02 by Avamander