nRF24/RF24Network #115: Problem with Attiny84

tguillaume02, 2017-10-02:
Hello,
I use your library RF24 and RF24Network.
I have four Arduino Uno and one attiny84 with NRF24L01+.
The Network is:
0 Arduino UNO
1 Arduino UNO
     011 Attiny84
        0111 Arduino UNO
     021  Arduno UNO

The probleme is with the Attiny84, when I send data to 011 from 0 sometimes ( often ) no data is received.
The same for 0111 from 0.

Do you have an idea ?
 
Thanks

Avamander, 2017-10-02:
Can't really debug your system for you. 

Start debugging the hardware level:
1) Check wiring, every. wire.
2) Check power, use a multimeter at minimum to make sure voltages are in normal range, oscilloscope to detect ripples or brownouts
3) Decouple the power lines (both input to the MCU and input of the nRFs) with an electrolytic capacitor (10000uF should be sufficient to get rid of low frequency noise, add a small ceramic to get rid of the high frequency noise)
4) Update your IDE and your libraries (RF24, RF24Network, boards, everything)
5) Enable debugging by commenting in certain preprocessor flags you can find in the library's files, it's all described there
6) Do not put nodes too close, high power transmission can be lost if nodes are too close
7) Dump EVERY node's logs to a file with a timestamp (also read 5. point again, a MUST if you want to debug it)
8) Now, if by this moment you can see it's actually the library and there's literally NO WAY that it's your setup describe the issue here in full detail, include all the logs you collected (pastebin for ex.). If the library is freezing, you can literally add a print statement after every line of the library's code to see where it freezes. Unfortunately cheaper AVRs do not have fancy hardware debuggers.

tguillaume02, 2017-10-02:
Thanks.
For the moment I have check : 
I have check every wire, all is ok.
I have check the power beetween GND and VCC of NRF24 I have 3.3V but after the nrf24 became hot.
I have add two electrolytic capactior (10µf and 0.1µf )
[image].

My IDE and Library are Update 
IDE V.1.8.4 ( windows store ) 
I can't debug attiny with serial.
All nodes are not in the same room.
I don't understant how make  

> Dump EVERY node's logs to a file with a timestamp (also read 2. point again, a MUST, if you want to debug it)

Thanks

Avamander, 2017-10-02:
> After nRF24 became hot
So you don't know if it's actually undamaged.
> I have add two electrolytic capactior (10µf and 0.1µf )
Both are tiny, not enough to filter out any significant ripples.
> I can't debug ATTiny with serial
Find a way to debug the ATTiny then.
> I don't understand how make
Collect the serial debug logs, make them have timestamps (when the lines were printed), can't really make this clearer.

tguillaume02, 2017-10-02:
> So you don't know if it's actually undamaged.
I have test the nRF24 On the Uno after and it works

> Both are tiny, not enough to filter out any significant ripples.
I have add an 4,7F between GND AND 3V
[image]

I have test debug. 
I print data when Master ( 0 ) send and when first child (011) transmit to her child.
And I print on (011 ) when ( 0111 ) have received value by transmitting the received value.
So with the result debug :  0 printed data send, 011 printed data send but 011 didn't received data from 0111

Avamander, 2017-10-02:
> 4.7F

Uhh, are you totally sure?

> I print data [...]

Enable the network's built-in debugging and pastebin the logs with timestamps, then I can actually look at what's going on, your description now doesn't tell me more than it tells you.

tguillaume02, 2017-10-02:
Sorry 4,7µf.
I have forgot µ.

Can you tell me where put capacitor 10000uF ( between 5v and GND ) and ceramic capacitor ( between 3.3v and GND ) ?

Can you tell me where or what flag activate for enable debugging.

Thanks

Avamander, 2017-10-02:
> Can you tell me where put capacitor 10000uF ( between 5v and GND ) and ceramic capacitor ( between 3.3v and GND ) ?

On the nRF24L01+

> Can you tell me where or what flag activate for enable debugging.

See RF24Network_config.h
```
    /** Debug Options */
    //#define SERIAL_DEBUG
    //#define SERIAL_DEBUG_MINIMAL
    //#define SERIAL_DEBUG_ROUTING
    //#define SERIAL_DEBUG_FRAGMENTATION
    //#define SERIAL_DEBUG_FRAGMENTATION_L2
    /*************************************/
```

tguillaume02, 2017-10-02:
Ok Thanks

But I'm not sure about understanding for capacitors :s

tguillaume02, 2017-10-03:
I have uncomment this:
   //#define SERIAL_DEBUG
    //#define SERIAL_DEBUG_MINIMAL
    //#define SERIAL_DEBUG_ROUTING
    //#define SERIAL_DEBUG_FRAGMENTATION
    //#define SERIAL_DEBUG_FRAGMENTATION_L2

But I have not print, I don't understand.

I have test send from attiny, no problem I recevied data send by Attiny.
Only problem for received data from other node on Attiny.

I have change capactior.
I put 200µF between 5V and GND of AMS1117
And 47µf between 3.3v and GND of nRF