nRF24/RF24Network #123: debug & active nodes

mmta41, 2018-02-20:
I tried to enable serial debug so uncomment the line
``#define SERIAL_DEBUG``

in RF24Network_config.h file but there is no debug output.
----
I'm trying to keep list of all active nodes in the network, so I ping whole network in a period of time,
but the `update`  method is also doing such thing, is it possible to get all active nodes in `update` method? or using single packet with `NETWORK_POLL` header?

Avamander, 2018-02-20:
> in RF24Network_config.h file but there is no debug output. (I did clean & build)

Make sure you initialize printf.

>  is it possible to get all active nodes in update method? 

Sure, if you write the code to do so.

mmta41, 2018-02-20:
@Avamander thanks for reply
since I'm new to c++ can you explain more about how initialize printf or give me a refrence, I'm currently on arduino board.

and about scanning network, I'm trying to say does update method getting list of all nodes? 
and is it possible to scan network using single packet with `NETWORK_POLL` header?

Avamander, 2018-02-20:
> since I'm new to c++ can you explain more about how initialize printf or give me a refrence,

`printf_begin();` in `setup()`

> I'm trying to say does update method getting list of all nodes?

Not exactly, see update() [here](https://github.com/nRF24/RF24Network/blob/master/RF24Network.cpp#L141) to understand what it does.

> and is it possible to scan network using single packet with NETWORK_POLL header?

Sure, if you write the code to do so. I recommend you read the documentation about what NETWORK_POLL is for:

> Messages of this type are used with multi-casting, to find active/available nodes. Any node receiving a NETWORK_POLL sent to a multicast address will respond directly to the sender with a blank message, indicating the address of the available node via the header.

closed 2019-05-10 by TMRh20