nRF24/RF24Network #129: network.write bug in interrupt

tguillaume02, 2018-05-22:
When network.write is call in interrupt, this failed and need to restart.

`const uint16_t this_node = 031;
const uint16_t other_node = 0;

#include <RF24.h>
#include <RF24Network.h>
#include <PinChangeInterrupt.h>

#define relay 9

void setup()
{
  radio.begin();
  radio.setDataRate(RF24_250KBPS);
  network.begin(this_node);
  
  attachPCINT(digitalPinToPCINT(1), changeState, CHANGE);
  pinMode(relay, OUTPUT);

  delay(1000);
  SendDataRF24(relay, 0, 0, other_node, 1);
}

void loop()
{
  network.update();  
}

void SendDataRF24(int pin, int value, int etat, int other_node,  int type = -99)
{
 payload_t payload = { pin, value, etat, type };

 RF24NetworkHeader header( other_node,  't');
 network.write(header, &payload, sizeof(payload));
}

void changeState()
{
  int val = !digitalRead(relay);
  digitalWrite(relay, val);
  SendDataRF24(relay, val * 100, val, other_node, 1);
}`

Avamander, 2018-05-22:
Set a flag inside the interrupt routine and later send the message in the main loop (and unset the flag).

tguillaume02, 2018-05-22:
Yes but I don't understand why is not work in Interrupt ?

closed 2019-05-10 by TMRh20