nRF24/RF24Network #130: Rpi perfomance

dimquea, 2018-06-25:
I made small programm for rpi 1. Programm read network, when message from other node available, then make mysql query. 
All code in the while(1) cycle, like in example. And I see 100% CPU usage every time, even idle. 
So how I can reduce the load, without packet losing?
Thank you.

wilmsn, 2018-06-25:
Even i don't know your sourcecode but as you are in a multitasking system you should always do something like this:
....
while(1) {
# do your job here
usleep(1000) #sleeptime in us. change it to whatever you can give the processor to other tasks
}

dimquea, 2018-06-25:
I think about same. But how about radio? What happens when message will be arrived, but usleep still work?

wilmsn, 2018-06-25:
Message will be buffered in the radio chip. You should choose a sleep time that makes sure that no messages go lost. Read the datasheet of nrf24l01 chips to get more information about build in registers.

dimquea, 2018-06-26:
i checked your rf24hub. very interesting. thank you.

closed 2019-05-10 by TMRh20