nRF24/RF24Network #131: ::available() causes Soft WDT reset on NodeMCU

JaredP94, 2018-08-18:
Hi there, running a NodeMCU board as a receiver in an RF24Network config. Using the network `available()` check seems to reset the board if it's not true.

Sources seem to indicate this should work for the ESP architecture but I'm not sure about that IMHO.

Can confirm that swapping NodeMCU and Arduino roles to TX and RX roles respectively works without issue. Any advice on the matter would be greatly appreciated!

Implemented receiver code:
```
#include <SPI.h>  
#include "RF24.h" 
#include "RF24Network.h"

#define NUM_PARKINGS 24
#define NUM_SENSORS 4
#define NUM_ROWS 3

RF24 myRadio (2, 15);
RF24Network network(myRadio);

bool received_results;
bool incoming;
int row_number;
signed char dataReceived[NUM_PARKINGS];
signed char parkingBayData[NUM_ROWS][NUM_PARKINGS];

// Octal Mapping
const uint16_t this_node = 04;
const uint16_t gateway_node = 00;

bool row1 = false;
bool row2 = false;
bool row3 = false;

void setup() { 
  Serial.begin(9600); 
  Serial.println(F("Transmitter - Node19")); 
  
  for (int i = 0; i < NUM_PARKINGS; i++) {
    dataReceived[i] = -1;
  }

  for (int i = 0; i < NUM_ROWS; i++) {
    for (int j = 0; j < NUM_PARKINGS; j++){
      parkingBayData[i][j] = 0;
    }
  }
  
  SPI.begin();
  myRadio.begin(); 
  network.begin(90, this_node);
  myRadio.setDataRate(RF24_250KBPS);
  myRadio.setPALevel(RF24_PA_MIN); 
} 

void loop() { 
  incoming = false;
  received_results = false;
  row1 = false;
  row2 = false;
  row3 = false;

  while(!received_results){
    network.update();

    if (network.available() == true || network.available() == false){
      incoming = network.available();
    }
    
    if (incoming){
      Serial.println("Recv");
      RF24NetworkHeader header;
      network.read(header, &dataReceived, sizeof(dataReceived)); // Read the incoming data
      Serial.print("Data received from admin: ");
      
      if (header.from_node == gateway_node && !row1){
        memcpy(parkingBayData[0], dataReceived, NUM_PARKINGS * sizeof(char));
        
        Serial.println("Row 1 from gateway: ");
        for (int k = 0; k < NUM_PARKINGS; k++) {
          Serial.print(parkingBayData[0][k]);
          Serial.print(" ");
        }
        Serial.println("");
        
        row1 = true; 
      }

      else if (header.from_node == gateway_node && !row2){
        memcpy(parkingBayData[1], dataReceived, NUM_PARKINGS * sizeof(char));
        
        Serial.println("Row 2 from gateway ");
        for (int k = 0; k < NUM_PARKINGS; k++) {
          Serial.print(parkingBayData[1][k]);
          Serial.print(" ");
        }
        Serial.println("");
        
        row2 = true;
      }

      else if (header.from_node == gateway_node && !row3){
        memcpy(parkingBayData[2], dataReceived, NUM_PARKINGS * sizeof(char));
        
        Serial.println("Row 1 from gateway ");
        for (int k = 0; k < NUM_PARKINGS; k++) {
          Serial.print(parkingBayData[2][k]);
          Serial.print(" ");
        }
        Serial.println("");
        
        row3 = true;
      }

      if (row1 && row2 && row3){
        received_results = true;
      }
    }
  }

  Serial.flush();

  delay(8000);
}
```

Avamander, 2018-08-18:
Are you on the latest version of ESP8266 boards file?

JaredP94, 2018-08-18:
Seems I was a bit naive in decoding the stack trace - the issue correlated to the loop which the `available()` command was allowing to occur. The WDT seems to have an indefinite loop reset period of 50 ms which prompted the hard resets.

Thanks to [https://github.com/esp8266/Arduino/issues/3680#issuecomment-376541271](this issue), I was able to find that putting a `delay(0)` at the end of the while loop seemingly fixed it 😄 

Sorry for the inconvenience!

closed 2018-08-18 by JaredP94