nRF24/RF24Network #133: [Question] Delay after network write when target is offline/unreachable

MaartenJB, 2018-12-16:
Hi,

I noticed when I do a network.write and the connection to my target is poor to none, the write function is delayed. (30 ms or so)

I have disabled AutoAck and Retries, but this does not change anything. Is the hardware always expecting an ack package and is this causing the delay?

Thanks, best regards,

Maarten

TMRh20, 2018-12-16:
The message type defines how it will be handled. User message types 1-64 will not be acknowledged, so that should minimize the delay.

MaartenJB, 2018-12-16:
Hi, I forgot to mention that I used message type 1.

TMRh20, 2018-12-16:
The hardware is always expecting an ack package. The user type only affects network acks, so I guess you are seeing the radio retry delay. This is configured in RF24Network.cpp in network begin. Try setting radio.setRetries(retryVar,0); and txTimeout to 1, or as preferred.

You could also use a direct write to a node in range which uses unicast. ( write(header, message, 1)  )

MaartenJB, 2018-12-16:
thanks, reducing the txTimeout speeds things up a lot.
with the unicast write function you mean this one right?

`

bool RF24Network::write | ( | RF24NetworkHeader & | header,
-- | -- | -- | --
  |   | const void * | message,
  |   | size_t | len
  | ) |  

`

TMRh20, 2018-12-16:
Shown here under advanced operation with writeDirect . http://tmrh20.github.io/RF24Network/classRF24Network.html

MaartenJB, 2018-12-16:
thanks a lot!

closed 2019-05-10 by TMRh20