nRF24/RF24Network #136: Cannot Send more than 24 bytes of data from arduino to raspberry pi

marcus011, 2019-05-15:
Hi,
I am trying to send data from Arduino to the raspberry pi using RF24Network library,
when I try to send a 32 packet size data then I got an error 

> Cannot enqueue multi-payload frames to self

same error receiving on python as well as on c++

channel using 115
& using them as broadcast so same id on TX node & RX node

I have also change receiver node but still same error

so can anyone tell how I can receive 32 bytes of data on raspberry side

TMRh20, 2019-05-15:
You need to have separate ids for each device. To broadcast, see the documentation for multicast and multicast relay.

marcus011, 2019-05-15:
Thanks a lot I will look into it.

closed 2019-05-15 by Avamander