nRF24/RF24Network #137: How to use multicast relay?

marcus011, 2019-05-31:
Hi, 
I need to broadcast message to same ID's, earlier I was using Rf24Network but having some issues with but as recommended by  @TMRh20 [https://github.com/nRF24/RF24Network/issues/136#issuecomment-492577582](url) to use multicast and multicast relay. But I have gone through the NRF Documents [http://tmrh20.github.io/RF24Network/index.html](url) and didn't find any tutorial or great explanation of how to use this.
So can you help me how to use the multicast relay for broadcasting?

Thanks

TMRh20, 2019-05-31:
If you take a look at the image [here](http://tmrh20.github.io/RF24Network/Tuning.html) , it shows the different 'levels' the network is divided into. 

For example, to send a broadcast message from the master node to all other nodes, just call [network.multicast()](http://tmrh20.github.io/RF24Network/classRF24Network.html#a25e19015dbf5073a8401e0fe71f68848) and send the data to 'level 1'

The nodes at level 1 (nodes 01 - 05) need to have multicast forwarding on if you want them to forward multicast messages over extended distances (to level 2, etc). For this, each node should have [multicast relay](http://tmrh20.github.io/RF24Network/classRF24Network.html#add7408ca2b2f9ab15331ba77a747f7bb) enabled.

Alternately, if all of the nodes are in range of the master, you can set them all to use the [same multicast 'level'](http://tmrh20.github.io/RF24Network/classRF24Network.html#a4fb7c61eee75d708a4ba20bb1caec54f) and no forwarding is required.

Hope that helps,

TMRh20

marcus011, 2019-06-03:
Thanks @TMRh20 for replying, right now my all device is having same id i.e 01 and after your response, I have used this.
`  bool ok = network.multicast(header, &rm_payload, sizeof(rm_payload), 01);`
so can you tell will this work properly & did all my nodes with id 01 will receive the message?
and I don't want my data to shift to another level, I need only one level.

TMRh20, 2019-06-04:
It might but technically you would want to set proper addressing and set the multicast level to one on each node per my previous message.

Cyberstorm11, 2020-04-11:
Hi,
I have a network of adunios all running the RF24network library.

I am also trying to use the multicast function on 2 layers of devices. 
00 -> 01 -> 011
               -> 021 etc
When I turn on multicast relaying on node 01 it stops responding to the broadcast message.

Master Code snippet:
```
//Send the message
  network.update();
  RF24NetworkHeader header(01);     // (Address where the data is going)
  bool ok = network.multicast(header, &outData, sizeof(outData), level); // Send the data
  delayMicroseconds(500);
  if (level == 01) level = 02;
  else level = 01;
```
nodes
`    //NRF24
    SPI.begin();
    radio.setRetries(5, 15);
    network.txTimeout = 486 + this_node * 7;
    radio.begin();
    network.begin(55, this_node); //(channel, node address)
    network.multicastRelay = this_node < 9;//only layer one nodes multicast
`
It appears on node 01 the network.available() does not return true once multicast is enabled.
What have I missed?

Thanks for any help.

TMRh20, 2020-04-12:
The issue is not apparent from your code. I would suggest turning on debugging (RF24Network_config.h) to get more info and double check ur code.

Cyberstorm11, 2020-04-19:
Thanks for the replay I am getting serial information and have attached a dump from node01Debug Serial.

My limited understandings that duplicate messages appear to be filling the message buffer?
[Node01Debugfile.txt](https://github.com/nRF24/RF24Network/files/4498545/Node01Debugfile.txt)

TMRh20, 2020-04-21:
That’s weird, looks like the multicast levels are messed up or something. I’m guessing it’s going to be something in your code. I haven’t gotten a chance to look closely at this but I’ll take a closer look as soon as I can.

Cyberstorm11, 2020-04-21:
Thanks for your time below are both ends of the code its a bit messy being in development.

Master
`
#include <RF24.h>
#include <RF24Network.h>
#include <SPI.h>
#include "printf.h"

RF24 radio(7, 8);                // nRF24L01 (CE,CSN)
RF24Network network(radio);      // Include the radio in the network
const uint16_t this_node = 00;   // Address of this node in Octal format ( 04,031, etc)
const uint16_t node01 = 01, node011 = 011, node021 = 021, node031 = 031;
const uint16_t node02 = 02, node012 = 012, node022 = 022, node032 = 032;
const uint16_t node03 = 03, node013 = 013;

const int Number_ch = 14;// Number of LED channels being controlled

bool Serial_EN = true;// Enable or disable serial communications

uint8_t level = 01;

const int blinkPin = 2;//Blink Indication Pin
bool toggle;

//Serial Variables
String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete

//timer setup
unsigned long currentTime = 0;

//Timer Structure
struct timer {
  unsigned long lastTime;
  unsigned long interval;
  bool output;
};
timer timers[10];

//Channel Structures
struct channel {
  byte pwm;//Pulse Width value of channel 0 to 255
  bool side1; //Side 1 enabled
  bool side2; //Side 2 enabled
  bool change;// has the pattern changed
  byte pattern;//pattern assignment
  bool rampUp;//Current ramp direction is rising
  timer pTimers[5];//pattern timers
};

//Channel Declaration
channel ch[Number_ch];

//Network Data Structure
struct netData {
  byte data1;//PWM channel 1
  byte data2;//PWM channel 2
  byte data3;//Command Byte
  /*Command Byte
     0 = 0 all off
     0 = 1 enable other functions
     1 = 1 channel 1 side 1 On
     2 = 1 channel 1 side 2 On
     3 = 1 channel 2 side 1 On
     4 = 1 channel 2 side 2 On
     5 reserved
     6 reserved
     7 reserved
  */
[... 509 lines cut ...]
          if (channel_2_s1 && channel_2_s2) { // If both sides are requested switch both regularly
            analogWrite(channel_2.PIN_1, 127);
          }
          else if (channel_2_s1) {
            analogWrite(channel_2.PIN_1, 255);
          }
          else if (channel_2_s2) {
            analogWrite(channel_2.PIN_1, 0);
          }
        }
        else analogWrite(channel_2.PIN_EN, 0);

      }
    }
    else if (rfCount > 61000){
      if (Serial_EN) Serial.println(rfCount);
     // resetFunc();  //call reset
    }
    else {
      analogWrite(channel_1.PIN_EN, 0);
      analogWrite(channel_2.PIN_EN, 0);
    }
  }
  // END OF LOOP
`

TMRh20, 2020-04-24:
Not sure what to say here, I get different results. The queue doesn't seem to fill up, and the forwarding appears to be working.  
  
I notice my debug output is messed up a bit and reporting incorrect values when running your code, so without analyzing further I'm guessing you have a memory issue or something going on. (overwriting a buffer or something)  
  
One thing you should change in your code is to not configure the radio or network until calling radio.begin and network.begin.

My best suggestion would be to comment out large portions of your code and re-test until you get something that works. Then build from there and keep track of where it goes bad.

closed 2020-05-27 by TMRh20