nRF24/RF24Network #138: How to route from grand-child to master

mixaobo, 2019-06-08:
i've succeeded with node00 and node01, now i want to test node 00 node01(router) and node011
*Node011:  //Don't have any function reference to node01
`network.update();
RF24NetworkHeader header(00);
network.write(header, &payload, sizeof(payload)); 
`
*Node01: ////Don't have any function reference to node011
`network.update();`
*Node00:
`network.update();
RF24NetworkHeader header;
network.read(header, &payload, sizeof(payload));
`
So this is my short code of my project, so it will automatically route to node00 (just make sure node01 in range within (node00 and node011)

wilmsn, 2019-06-08:
How is your var header set when you try to send a message from node 00 to 011 or back?

mixaobo, 2019-06-08:
> How is your header set when you try to send a message from node 00 to 011 or back?
> Make sure that to_node (inside the header) is set to the final destination node.

i see. first i have node011 and noded00 but cant receive payload.
then i deploy node01 (just ust network.update()) then everything is ok!

closed 2019-07-29 by mixaobo