nRF24/RF24Network #139: How to use RF24network sleep mode in ESP32

mixaobo, 2019-06-25:
I' m doing RF24network in ESP32. I'm only do with esp32 now (not download any arduino or esp8266 library) and have trouble with sleep_mode. i have uncomment endable_sleep_mode in config file and got this `C:\Users\Admin\Documents\Arduino\libraries\RF24Network\RF24Network.cpp:35:24: fatal error: avr/sleep.h: No such file or directory`
i have downloaded avr-libc-master but dont know where should i put this library.
Thank you

TMRh20, 2019-06-25:
The built in sleep mode only works with AVR devices, with ESP, you need to use ESP specific sleep modes. See 
https://www.google.ca/amp/s/lastminuteengineers.com/esp32-sleep-modes-power-consumption/amp/

mixaobo, 2019-06-25:
I see. In RF24.h, there are powerup() and powerdown() function. Do i have to use it. Like
Powerdown();
Esp sleep like you post above;
Pơerup()

mixaobo, 2019-06-25:
> The built in sleep mode only works with AVR devices, with ESP, you need to use ESP specific sleep modes. See https://www.google.ca/amp/s/lastminuteengineers.com/esp32-sleep-modes-power-consumption/amp/
> […](#)

I see. In RF24.h, there are powerup() and powerdown() function. Do i have to use it. Like
Powerdown();
Esp sleep like you post above;
Powerup();

TMRh20, 2019-06-25:
Uhh, no you have to use ESP specific code. Google for ESP32 low power modes.

mixaobo, 2019-06-25:
> Uhh, no you have to use ESP specific code. Google for ESP32 low power modes.
> […](#)

Thank you. Close issue

closed 2019-06-25 by mixaobo