nRF24/RF24Network #144: RF24Network with struct's

TKBeckum, 2019-08-06:
I am trying to build a RF24Network with three nodes (00, 01, 02). All three have to send and receive a struct. However these structs are all different.
In most examples they use 'unsigned long incomingData' in the 'network-read' statement. But that can hold an integer only, I suppose.

Does anyone know how to write the 'network.read' statement in this case?
Can I for example first read the header to see from which node the message is received ( with a statement like 'if (header.from_node == 1)'), because than I know which struct is in the incoming message.

wilmsn, 2019-08-06:
If you use different structures for you payload, you can use the "peek()" function to read the network header, use the right payload structure and then use read().
Have a look at the docs for details: http://tmrh20.github.io/RF24Network/classRF24Network.html

TKBeckum, 2019-08-07:
Thanks very much! I'll give it a try.
One more question, if you allow me. In send examples 'RF24NetworkHeader header2(node01);'is used. Therefor I assumed this always has to be the address of the message. However in a receive example I see: 
     network.peek(header); if(header.type == 'T'){}
This suggests that the header can be something else than the node address. Any suggestion where I can find more on this?
Thanks in advance

wilmsn, 2019-08-07:
Take a look at http://tmrh20.github.io/RF24Network/structRF24NetworkHeader.html
Use "to_node" for the address/receiver of the data. Address has to be in octal like "01" for node 1 and 021 for sec. child of node 1. The type you can use free as you need it - see restriction in the doc.
I did a lot with it in my github.com/wilmsn/rf24hub - examples for arduino are included. Feel free to take a look there.

TKBeckum, 2019-08-07:
Thanks Norbert! I hope I will be able to understand fully the examples you
present on GitHub. I am a novice.
But the "from_node" and "to_node" I understand and can use.
One thing I am confused about is the octal addresses used in the examples:
The base-node or node-0 is "00". But why is node-1 "01" and not "10" (the
first child of node 0) like 021 for sec. child of node 1? Or is this
because node-0 is always there thus can be 'skipped' in the address?

wilmsn, 2019-08-07:
For addressing see: http://tmrh20.github.io/RF24Network_Dev/Addressing.html
As long as you only use node 0...3 you can leave the leadimg 0 away ( 02 = 2 )!  The leading zero tells the compiler that the number is octal (=3 bits):

closed 2019-08-08 by TKBeckum