nRF24/RF24Network #150: Receiver stops receiving and Sending after one time communication happens...!

hanishgopi, 2019-11-15:
My receiver stops and gets no data after receiving and sending once....and also resets the Arduino sometimes

I have enabled #define failure handler in RF24_config.h
I m running this chip on channel 90
I m using a SIM800A GSM module connected to this receiver.
please help me to solve this problem.... @TMRh20
void setup() {

Serial.begin(9600);
Serial.println("Wait...");

//Initialize GSM module
SerialAT.begin(9600);
setup_internet();
getDateTime();
delay(2000);
Serial.println("Intializing Radio module:");
SPI.begin();
radio.begin();
network.begin(90, Master_node);
radio.enableDynamicPayloads();
radio.setDataRate(RF24_250KBPS);
Serial.println("Master program Started");
Serial.println(TIME);

}

void loop()
{

if (network.available())
{
RF24NetworkHeader header;
network.read(header, &slave_data, sizeof(slave_data));
Serial.println("Received Data from NODE: ");
Nodemsg = String(slave_data[0]) + String(slave_data[1]);
Serial.println(slave_data);

}
else if (Nodemsg == "T1")
{
getDateTime();
RF24NetworkHeader header(Slave_1_node);
network.write(header, &TIME, sizeof(TIME));
Serial.println("Sent data");
Serial.println(TIME);
Nodemsg = "";
}
else {
network.update();
for (int i = 0 ; i < 91 ; i++)
{
slave_data[i] = '\0';
}
}
}

Avamander, 2019-11-15:
Please try running unmodified examples (only change the roles and pin mapping if really required), do they work for a prolonged period of time?

The most probably issue is that your Arduino's 3.3V regulator is too weak to feed both the GSM module and the nRF, you'd probably either need a separate step-down module or a beefy linear regulator.

hanishgopi, 2019-11-15:
> Please try running unmodified examples (only change the roles and pin mapping if really required), do they work for a prolonged period of time?
> 
> The most probable issue is that your Arduino's 3.3V regulator is too weak to feed both the GSM module and the nRF, you'd probably either need a separate step-down module or a beefy linear regulator.

No im using separate power supply for gsm module and i have tried with few brand new Arduinos also...

Avamander, 2019-11-15:
You can find the examples in the examples folder of the library, also quickly accessible trough Arduino IDE's menu.

If you get Arduino resets then there's either grave mistakes in your code that aren't related to this library or there isn't enough power. Decoupling the 3.3V rail of the Arduino is a good idea in all cases though.

hanishgopi, 2019-11-15:
> You can find the examples in the examples folder of the library, also quickly accessible trough Arduino IDE's menu.
>
> If you get Arduino resets then there's either grave mistakes in your code that [...]

i face this problem only when receiver as Arduino and transmitter as barebones atmega328p

Avamander, 2019-11-16:
One of the grave mistakes in your code might be running our of RAM, you have to check for it.

closed 2020-08-16 by TMRh20