nRF24/RF24Network #153: Repeater Possible?

marcus011, 2020-02-26:
I need to ask, is it possible to create Repeater ?

My one device is at id 011
My gateway or end device is at 00
Now i need to create multiple repeater with same id i.e 01 so that data can travel both ways.
Is it possible with this ?

Basically my aim is to transmit data from one end to other but the distance is long so for that i need to use nrf as a repeater in between like we ZigBee does.

Please let me know of its possible or not ? Or any another way i can achieve this.

TMRh20, 2020-02-26:
RF24Network nodes will automatically act similar to zigbee repeaters as long as your network is configured correctly. 

For example, as long as node 01 is online, it will forward all traffic addressed to children 011-041 from node0 (master node) and vice-versa, traffic from children 011-041 will route through node 01 to get to master.

All you need to do is address your message to the end node, and nodes in between will relay the messages.

See https://tmrh20.github.io/RF24Network/Tuning.html for a bit more information on topology and routing.

marcus011, 2020-02-26:
Thanks for your reply @TMRh20 .
I have already gone through this, but what i want is communication between 100+ devices to multiple master nodes through repeaters,so for that i want repeater like device so that range also increase plus multiple devices can through there data .
So can you help with any more suggestions.

marcus011, 2020-02-26:
Can we make multiple devices with same ids like 01 to all & when we need to push data we will will push that data on 01 & 01 device will push data forward till reach the end device.

TMRh20, 2020-02-26:
Hmm, well you need to consider the limitations of the radios if looking at using 100+ devices. Typically, 15 to 25 radios with a single master, and regular communication ( data sends every few seconds or so) is reasonable. Depending on traffic requirements, the network can handle more or less nodes.

If you want to use multiple masters and multiple routing nodes, you would need to configure them on separate radio channels. Ie: Nodes 0 and 01 configured on channel 1, additional nodes 0 and 01 configured on channel 20. Child nodes 011-041 can send using channel 1, change the radio channel, then send again using channel 20. If you want bidirectional communication, you probably would need to have the master on each channel respond to the nodes when they send data.

That is about the closest I can think of to what you are wanting to achieve.

marcus011, 2020-02-26:
Thanks @TMRh20  for your suggestions,but my requirement is not yet satisfied. So I will try something & let you know if any more help required.
✌🏻

closed 2020-03-07 by marcus011